AdnanHodzic/displaylink-debian #485: Displaylink Kali Linux

volkpa99, 2020-10-29:
Hello,

I have the HP USB-C Universal Dock.
With Windows it works without problems and everything is transferred. 
Now I have a Linux computer Thinkpad T450s, Kali Linux (5.7.0-kali1-amd64, Release: 2020.3) is installed.

The mouse and keyboard, ethernet is transferred, unfortunately only no display on the external monitors. (Monitors remain black)
In addition I read that the DisplayLink driver must be installed?
If I install it successfully via your script(docking station is connected) then the computer will not start properly if the docking station is not connected.
The PC then hangs in boot mode, only if the docking station is connected, the PC will boot up to the login screen.

Can you help me? Thanks for the support

Greetings

volkpa99, 2020-10-31:
root@kali:/home/volkpa/Downloads/DisplayLink/displaylink-debian# ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: Datei oder Verzeichnis nicht gefunden
root@kali:/home/volkpa/Downloads/DisplayLink/displaylink-debian# xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
root@kali:/home/volkpa/Downloads/DisplayLink/displaylink-debian#

what am I doing wrong?

Barabazs, 2020-10-31:
Can you run the install script and post the output here?
The install did not succeed, hence the debug error.

volkpa99, 2020-11-04:
Hi,
I have downloaded and installed everything again. After that it worked, the issue can be closed.
Thanks a lot!

closed 2020-11-04 by Barabazs