AdnanHodzic/displaylink-debian #488: Errors while installing

engineer22, 2020-11-03:
I'm on Debian 10 Buster, and while I run the script I get these errors. How can I solve them?[image]

Barabazs, 2020-11-03:
Please post the output of the install and debug please.

mdmle, 2020-11-03:
I have had the exact same issue since kernel 5.9 update on Debian testing (Bullseye). 

Digging a bit in the code, it seemed that the patch used, coming from https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch, isn't up-to-date. 
Thus, I've modified the following line in evdi.sh file
```
#patchFileName="evdi-all-in-one-fixes.patch"
patchFileName="evdi-kernel-5.9-test.patch"

```
And it's working as a charm. At least in my case.

Thanks a lot for providing this script !! 
Hope I can bring my little help ...

engineer22, 2020-11-03:
@Barabazs hello. I cannot post a debug, since this is within the installation process.

I can finish the installation successfully, but I spotted these errors now.

I try to install for months now, without success (I have a previous Issue) and now I spotted the errors.

engineer22, 2020-11-03:
@tchimper hey, thanks for the help.

I'm on Kernel 4.19, do you have any info on how I should apply the patch?

Where should I add the lines you mentioned?

Barabazs, 2020-11-03:
> I have had the exact same issue since kernel 5.9 update on Debian testing (Bullseye).
>
> Digging a bit in the code, it seemed that the patch used, coming from [...]

Thanks for confirming that patch worked for you. Could you post your debug output please?
That would help me to update displaylink-debian.

Barabazs, 2020-11-03:
> @Barabazs hello. I cannot post a debug, since this is within the installation process.
> 
> I can finish the installation successfully, but I spotted these errors now.
> 
> I try to install for months now, without success (I have a previous Issue) and now I spotted the errors.
> 
> Should I let the installer to finish and post the whole log?

Let the installer finish and then copy everything that was printed to your terminal and post it here.

You should still try to run the --debug procedure and also post the output here.

closed 2020-11-11 by engineer22