AdnanHodzic/displaylink-debian #489: Support for PopOS! 20.10 Groovy

JeroenGoddijn, 2020-11-04:
Here's the output when running `sudo ./displaylink-debian.sh --debug`
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
```

Barabazs, 2020-11-06:
Can you give me the output of:
`lsb_release -is`
`lsb_release -cs`

JeroenGoddijn, 2020-11-06:
> Can you give me the output of:
> `lsb_release -is`
> `lsb_release -cs`

Pop
groovy

Barabazs, 2020-11-06:
Can you try with this please:
`git clone https://github.com/Barabazs/displaylink-debian.git && cd displaylink-debian/ && sudo ./displaylink-debian.sh
`

JeroenGoddijn, 2020-11-06:
> Can you try with this please:
> `git clone https://github.com/Barabazs/displaylink-debian.git && cd displaylink-debian/ && sudo ./displaylink-debian.sh `

OK. But I already tried this yesterday:
```
wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh 
wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink.sh 
wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/evdi.sh
```
And then ran the displaylink-debian.sh, without any result...

 But I've been reading about more people having issues with 20.10, so it might be a DisplayLink issue and in 20.04 it supposedly works...

I'll try your command again

JeroenGoddijn, 2020-11-06:
Performing post install steps

Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xe" for details.

Barabazs, 2020-11-06:
Can you post the complete output of the installation?

And also the output of `sudo ./displaylink-debian.sh --debug`

JeroenGoddijn, 2020-11-06:
> sudo ./displaylink-debian.sh --debug

Here's the full output of installation:
```
jeroengoddijn@godsyogax1:~/Downloads/displaylink-debian$ sudo ./displaylink-debian.sh

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.8.0-7625-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-11-06 10:16:34--  https://www.displaylink.com/downloads/file?id=1576
Resolving www.displaylink.com (www.displaylink.com)... 34.250.82.133, 18.203.121.10, 52.214.244.113, ...
Connecting to www.displaylink.com (www.displaylink.com)|34.250.82.133|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1604679995&Signature=%2B02r83ZH5Rxxy0SPqg4xxaeSbpA%3D [following]
--2020-11-06 10:16:35--  https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1604679995&Signature=%2B02r83ZH5Rxxy0SPqg4xxaeSbpA%3D
Resolving s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)... 52.218.105.90
Connecting to s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)|52.218.105.90|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13889958 (13M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_5.3.1.zip’

DisplayLink_Ub 100%  13.25M  1023KB/s    in 14s         

2020-11-06 10:16:50 (965 KB/s) - ‘DisplayLink_Ubuntu_5.3.1.zip’ saved [13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install
[... 138 lines cut ...]
[[ Installing EVDI library ]]
make: Nothing to be done for 'default'.
[ Installing x64-ubuntu-1604/DisplayLinkManager ]
[ Installing libraries ]
[ Installing firmware packages ]
[ Installing licence file ]
[ Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager sctripts ]

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu

Installation complete!
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xe" for details.
```

JeroenGoddijn, 2020-11-06:
> sudo ./displaylink-debian.sh --debug

Here's the full output of this debug command:
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.8.0-7625-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.7.0.backup
5.1.26
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: 620
Integrated
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Barabazs, 2020-11-06:
Did you perform a reboot since trying the installation?
What's the output of `dkms status`?

JeroenGoddijn, 2020-11-06:
> dkms status

No haven't rebooted yet since the last install attempt, since it threw an error.

but here's the output of dkms status.
```
evdi, 1.7.0, 5.8.0-7625-generic, x86_64: installed
evdi, 1.7.0.backup, 5.8.0-7625-generic, x86_64: built
evdi, 5.1.26: added
system76, 1.0.9~1597073326~20.10~5b01933, 5.8.0-7625-generic, x86_64: installed
```

Barabazs, 2020-11-06:
At least evdi was installed correctly.

Can you reboot and try the reinstallation? And also post the output please.

JeroenGoddijn, 2020-11-06:
Just rebooted, ran the command again and selected R for reinstall... however, it seems Re-install just UNinstalls instead:
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: r

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.8.0-7625-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

DisplayLink Linux Software 5.3.1.34 install script called: uninstall
Distribution discovered: Pop!_OS 20.10

Uninstalling

[ Removing EVDI from kernel tree, DKMS, and removing sources. ]
dkms remove evdi/1.7.0 --all

-------- Uninstall Beginning --------
Module:  evdi
Version: 1.7.0
Kernel:  5.8.0-7625-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

evdi.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-7625-generic/kernel/drivers/gpu/drm/evdi/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.7.0
completely from the DKMS tree.
------------------------------
Done.
rm -f -rf /usr/src/evdi-1.7.0
/home/jeroengoddijn/Downloads/displaylink-debian
Stopping displaylink-driver systemd service
Removed /etc/systemd/system/multi-user.target.wants/displaylink-driver.service.
[ Removing suspend-resume hooks ]
[ Removing udev rule ]
[ Removing Core folder ]

Uninstallation steps complete.

-------------------------------------------------------------------

Performing clean-up
```

JeroenGoddijn, 2020-11-06:
So, I had to run `sudo ./displaylink-debian.sh` again and select "Install" which resulted in following output:
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.8.0-7625-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-11-06 11:10:48--  https://www.displaylink.com/downloads/file?id=1576
Resolving www.displaylink.com (www.displaylink.com)... 34.250.82.133, 34.252.69.235, 52.214.244.113, ...
Connecting to www.displaylink.com (www.displaylink.com)|34.250.82.133|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1604683249&Signature=Zouj1e6wsqkfojvlKJvEbi9jhqg%3D [following]
--2020-11-06 11:10:49--  https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1604683249&Signature=Zouj1e6wsqkfojvlKJvEbi9jhqg%3D
Resolving s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)... 52.218.29.219
Connecting to s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)|52.218.29.219|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13889958 (13M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_5.3.1.zip’

DisplayLink_Ubuntu_5.3.1.zip                    100%[=====================================================================================================>]  13.25M  8.80MB/s    in 1.5s    

2020-11-06 11:10:51 (8.80 MB/s) - ‘DisplayLink_Ubuntu_5.3.1.zip’ saved [13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
[... 134 lines cut ...]
[[ Installing EVDI library ]]
make: Nothing to be done for 'default'.
[ Installing x64-ubuntu-1604/DisplayLinkManager ]
[ Installing libraries ]
[ Installing firmware packages ]
[ Installing licence file ]
[ Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager sctripts ]

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu

Installation complete!
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Job for displaylink-driver.service failed because the control process exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xe" for details.
```

JeroenGoddijn, 2020-11-06:
After reboot the output of `dkms status` is: 

```
evdi, 1.7.0, 5.8.0-7625-generic, x86_64: installed
evdi, 1.7.0.backup, 5.8.0-7625-generic, x86_64: built
evdi, 5.1.26: added
system76, 1.0.9~1597073326~20.10~5b01933, 5.8.0-7625-generic, x86_64: installed
```

JeroenGoddijn, 2020-11-06:
`journal -xe` outputs: 
```
Nov 06 11:16:29 godsyogax1 systemd[1]: displaylink-driver.service: Scheduled restart job, restart counter is at 43.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ Automatic restarting of the unit displaylink-driver.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Nov 06 11:16:29 godsyogax1 systemd[1]: Stopped DisplayLink Driver Service.
░░ Subject: A stop job for unit displaylink-driver.service has finished
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A stop job for unit displaylink-driver.service has finished.
░░ 
░░ The job identifier is 5871 and the job result is done.
Nov 06 11:16:29 godsyogax1 systemd[1]: Starting DisplayLink Driver Service...
░░ Subject: A start job for unit displaylink-driver.service has begun execution
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit displaylink-driver.service has begun execution.
░░ 
░░ The job identifier is 5871.
Nov 06 11:16:29 godsyogax1 sh[8743]: modprobe: ERROR: could not insert 'evdi': Exec format error
Nov 06 11:16:29 godsyogax1 kernel: evdi: version magic '5.8.14 SMP mod_unload ' should be '5.8.0-7625-generic SMP mod_unload '
Nov 06 11:16:29 godsyogax1 sh[8769]: Module evdi/1.7.0 already installed on kernel 5.8.0-7625-generic/x86_64
Nov 06 11:16:29 godsyogax1 sh[8744]: modprobe: ERROR: could not insert 'evdi': Exec format error
Nov 06 11:16:29 godsyogax1 systemd[1]: displaylink-driver.service: Control process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ An ExecStartPre= process belonging to unit displaylink-driver.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Nov 06 11:16:29 godsyogax1 systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ The unit displaylink-driver.service has entered the 'failed' state with result 'exit-code'.
Nov 06 11:16:29 godsyogax1 kernel: evdi: version magic '5.8.14 SMP mod_unload ' should be '5.8.0-7625-generic SMP mod_unload '
Nov 06 11:16:29 godsyogax1 systemd[1]: Failed to start DisplayLink Driver Service.
░░ Subject: A start job for unit displaylink-driver.service has failed
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit displaylink-driver.service has finished with a failure.
░░ 
░░ The job identifier is 5871 and the job result is failed.
```

JeroenGoddijn, 2020-11-06:
Just rebooted again...this time with docking station plugged in and still nothing working...
not running any other commands until I get new instructions from you

Barabazs, 2020-11-06:
> Thanks for your responsiveness, by the way, @Barabazs
> Really appreciate it... Setup was working just fine, but upgrade to 20.10 killed my 2nd external monitor that is powered by docking station with DisplayLink. 

No problem. I've been in the same situation and can totally relate.

It seems the issue is that the module is being compiled against kernel 5.8.14, but you're actually running on kernel 5.8.0-7625-generic. I'm not sure how to help you with that.

You could try booting with another kernel version and retry the install procedure.

JeroenGoddijn, 2020-11-06:
> No problem. I've been in the same situation and can totally relate.
>
> It seems the issue is that the module is being compiled against kernel 5.8.14, but you're actually running on kernel [...]

So. how do I do that? No experience whatsoever with kernel customization or selection...
Don't want to break anything else...

Barabazs, 2020-11-06:
You don't have to customise anything. Just reboot and press space until you get the boot menu. You should be able to select another kernel version there.

JeroenGoddijn, 2020-11-06:
You're the man!!!
It's f***ing stupid, but loading with the old kernel restored my screen/docking and everything is working as normal again.

Why is this even a thing? That something works with an older kernel, but not with the newer one... *sigh

closed 2020-11-06 by Barabazs