AdnanHodzic/displaylink-debian #493: Debian 18.04

tempoeric, 2020-11-10:
Hi I have a problem on displaylink that was pulled on this repo. The problem occurs on the third monitor which was a portrait screen when i closed the lid of the laptop the second screen which is the portrait one still lags. What should i do to fix this

tempoeric, 2020-11-10:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.4.0-52-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Haswell-ULT
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): NVIDIA Corporation GM108M [GeForce 840M] (rev a2)
X11 version: 1.20.8-2ubuntu2.2~18.04.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2af cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x28d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x1de cap: 0x2, Sink Output crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
Provider 4: id: 0x26b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x242 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Barabazs, 2020-11-10:
Unfortunately not much...
It's a known issue. More info here: https://github.com/AdnanHodzic/displaylink-debian/issues/393#issuecomment-687780756

closed 2020-11-10 by Barabazs