AdnanHodzic/displaylink-debian #496: Question about evdi.sh

Barabazs, 2020-11-13:
Hi @AdnanHodzic 

Because there is currently no need to apply patches, other than the one to temporarily prevent reboot, I'm looking to refactor evid.sh. There is one line that I do not understand and would like to make sense to me before I proceed.

First we apply the patch and then sed....?
```shell
patch -Np1<$finalPatchPath
sed -E -e 's:SUBDIRS=([^ ]+) :M=\1 &:g' -i 'module/Makefile'
```

AdnanHodzic, 2020-11-13:
It seems to be modifying part of the patch, but I'm not 100% sure. I see these changes were introduced by you as part of #398

Barabazs, 2020-11-13:
It was actually introduced a while before #398 in this commit: https://github.com/AdnanHodzic/displaylink-debian/commit/6ea86231dd902db6a35b87395835fe3fcd34c160#diff-e28561ea1d1fa6a9e499eb81709e2a78b53a0b3b1886b8c23d1c177bda55acc3

I suppose it was necessary for the patch and only for that specific patch. I should read up on the usage of sed...

closed 2020-11-13 by Barabazs

AdnanHodzic, 2020-11-14:
I suggest you just run the same comment against that same patch file, as in this case sed will first output what it will do before doing it.