AdnanHodzic/displaylink-debian #497: install fails on ubuntu 20.10  - patch error

mayeulk, 2020-11-13:
Apparently a patch error on Ubuntu 20.10:
```
2020-11-13 12:06:22 (21,9 MB/s) - «evdi-all-in-one-fixes.patch» enregistré [6947/6947]

Clonage dans 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 9 (delta 0), pack-reused 0
Dépaquetage des objets: 100% (46/46), 66.98 Kio | 540.00 Kio/s, fait.
Note : basculement sur 'dc595db636845aef39490496bc075f6bf067106c'.

Vous êtes dans l'état « HEAD détachée ». Vous pouvez visiter, faire des modifications
expérimentales et les valider. Il vous suffit de faire un autre basculement pour
abandonner les commits que vous faites dans cet état sans impacter les autres branches

Si vous voulez créer une nouvelle branche pour conserver les commits que vous créez,
il vous suffit d'utiliser l'option -c de la commande switch comme ceci :

  git switch -c <nom-de-la-nouvelle-branche>

Ou annuler cette opération avec :

  git switch -

Désactivez ce conseil en renseignant la variable de configuration advice.detachedHead à false

patching file .travis.yml
Hunk #1 FAILED at 26.
1 out of 1 hunk FAILED -- saving rejects to file .travis.yml.rej
patching file Makefile
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
patching file library/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file module/Makefile
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
patching file module/evdi_drv.c
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
patching file module/evdi_drv.h
Hunk #1 FAILED at 20.
Hunk #2 FAILED at 51.
Hunk #3 FAILED at 112.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
patching file module/evdi_gem.c
Hunk #1 FAILED at 61.
Hunk #2 FAILED at 254.
Hunk #3 FAILED at 569.
Hunk #4 FAILED at 581.
4 out of 4 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1] : on entre dans le répertoire « /home/mk/bin/displaylink/evdi/module »
make -C /lib/modules/5.8.0-28-generic/build M=$PWD
make[2] : on entre dans le répertoire « /usr/src/linux-headers-5.8.0-28-generic »
  AR      /home/mk/bin/displaylink/evdi/module/built-in.a
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_drv.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_modeset.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_connector.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_encoder.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_main.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_fb.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_gem.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_painter.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_params.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_cursor.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_debug.o
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi_ioc32.o
  LD [M]  /home/mk/bin/displaylink/evdi/module/evdi.o
  MODPOST /home/mk/bin/displaylink/evdi/module/Module.symvers
  CC [M]  /home/mk/bin/displaylink/evdi/module/evdi.mod.o
  LD [M]  /home/mk/bin/displaylink/evdi/module/evdi.ko
make[2] : on quitte le répertoire « /usr/src/linux-headers-5.8.0-28-generic »
make[1] : on quitte le répertoire « /home/mk/bin/displaylink/evdi/module »
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1] : on entre dans le répertoire « /home/mk/bin/displaylink/evdi/library »
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc 
cp libevdi.so.1.7.0 libevdi.so
make[1] : on quitte le répertoire « /home/mk/bin/displaylink/evdi/library »
./displaylink-debian.sh: ligne 458: /home/mk/bin/displaylink/resources/displaylink-installer.patch: Aucun fichier ou dossier de ce type
```
Lat line "Aucun fichier ou dossier de ce type" means the file or folder is missing. I launched the script from /home/mk/bin/displaylink/ 
In effect, thers is no displaylink-installer.patch file in the above-mentioned folder
List of folders and files at this stage (tree command):
```
~/bin/displaylink$ tree
.
├── 5.3.1
│   ├── displaylink-driver-5.3.1
│   │   ├── 3rd_party_licences.txt
│   │   ├── arm-linux-gnueabihf
│   │   │   ├── DisplayLinkManager
│   │   │   └── libusb-1.0.so.0.1.0
│   │   ├── displaylink-installer.sh
│   │   ├── ella-dock-release.spkg
│   │   ├── evdi.tar.gz
│   │   ├── firefly-monitor-release.spkg
│   │   ├── LICENSE
│   │   ├── ridge-dock-release.spkg
│   │   ├── udev-installer.sh
│   │   ├── x64-ubuntu-1604
│   │   │   ├── DisplayLinkManager
│   │   │   ├── libevdi.so
│   │   │   └── libusb-1.0.so.0.1.0
│   │   └── x86-ubuntu-1604
│   │       ├── DisplayLinkManager
│   │       ├── libevdi.so
│   │       └── libusb-1.0.so.0.1.0
│   └── displaylink-driver-5.3.1.34.run
├── displaylink-debian.sh
├── displaylink.sh
├── DisplayLink_Ubuntu_5.3.1.zip
├── evdi.sh
└── resources
    ├── eaiof.patch
    └── evdi-all-in-one-fixes.patch

6 directories, 23 files

```

mayeulk, 2020-11-13:
Apparently this seems to be an issue with the wget way of installation described at https://github.com/AdnanHodzic/displaylink-debian#download-and-run-script-without-git-method-2 . Although it started well:
```
 wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh
--2020-11-13 11:11:35--  https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh
Résolution de raw.githubusercontent.com (raw.githubusercontent.com)… 151.101.120.133
Connexion à raw.githubusercontent.com (raw.githubusercontent.com)|151.101.120.133|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 24760 (24K) [text/plain]
Enregistre : «displaylink-debian.sh»

displaylink-debian.sh                            100%[=========================================================================================================>]  24,18K  --.-KB/s    ds 0,02s   

2020-11-13 11:11:35 (1,14 MB/s) - «displaylink-debian.sh» enregistré [24760/24760]

mk@mk-Latitude-7400:~/bin/displaylink$ wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink.sh
--2020-11-13 11:11:43--  https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink.sh
Résolution de raw.githubusercontent.com (raw.githubusercontent.com)… 151.101.120.133
Connexion à raw.githubusercontent.com (raw.githubusercontent.com)|151.101.120.133|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 701 [text/plain]
Enregistre : «displaylink.sh»

displaylink.sh                                   100%[=========================================================================================================>]     701  --.-KB/s    ds 0s      

2020-11-13 11:11:44 (7,51 MB/s) - «displaylink.sh» enregistré [701/701]

mk@mk-Latitude-7400:~/bin/displaylink$ wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/evdi.sh
--2020-11-13 11:11:51--  https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/evdi.sh
Résolution de raw.githubusercontent.com (raw.githubusercontent.com)… 151.101.120.133
Connexion à raw.githubusercontent.com (raw.githubusercontent.com)|151.101.120.133|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 1925 (1,9K) [text/plain]
Enregistre : «evdi.sh»

evdi.sh                                          100%[=========================================================================================================>]   1,88K  --.-KB/s    ds 0s      

2020-11-13 11:11:51 (5,66 MB/s) - «evdi.sh» enregistré [1925/1925]

mk@mk-Latitude-7400:~/bin/displaylink$ chmod +x displaylink-debian.sh evdi.sh
mk@mk-Latitude-7400:~/bin/displaylink$ sudo ./displaylink-debian.sh

```
Using instead the git way ( https://github.com/AdnanHodzic/displaylink-debian#repo-clone-method-1 ) there is no error message about missing patch file.
(I still need to reboot to see if it worked).
I'd suggest deleting the "without git" method in the doc, since in any case, using the wget method, the scripts at some  point asks to install git anyway. 
Mayeul

Barabazs, 2020-11-13:
Thank you for reporting this and the suggestion.

I'm currently in the process of refactoring some parts of displaylink-debian and will add this to my list.

closed 2020-11-13 by Barabazs