AdnanHodzic/displaylink-debian #499: hi, when i am installing on debian 10 i get this

MartinEtcheverry, 2020-11-13:
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at:
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-11-13 07:49:01--  https://www.displaylink.com/downloads/file?id=1576
Resolviendo www.displaylink.com (www.displaylink.com)... 52.214.244.113, 34.252.69.235, 18.203.121.10, ...
Conectando con www.displaylink.com (www.displaylink.com)[52.214.244.113]:443... conectado.
Petición HTTP enviada, esperando respuesta... 303 See Other
Localización: https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1605265142&Signature=mNZpIyaP9j96g48hC2Z6Cr4JQkY%3D [siguiendo]
--2020-11-13 07:49:03--  https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1605265142&Signature=mNZpIyaP9j96g48hC2Z6Cr4JQkY%3D
Resolviendo s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)... 52.218.108.211
Conectando con s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)[52.218.108.211]:443... conectado.
Petición HTTP enviada, esperando respuesta... 200 OK
Longitud: 13889958 (13M) [application/octet-stream]
Grabando a: “DisplayLink_Ubuntu_5.3.1.zip”

DisplayLink_Ubuntu_5.3.1.zip 100%[===========================================>]  13,25M  3,12MB/s    en 4,2s

2020-11-13 07:49:08 (3,12 MB/s) - “DisplayLink_Ubuntu_5.3.1.zip” guardado [13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%

-------------------------------------------------------------------
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias
Leyendo la información de estado... Hecho
libelf-dev ya está en su versión más reciente (0.176-1.1).
libdrm-dev ya está en su versión más reciente (2.4.97-1).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 1 no actualizados.
find: paths must precede expression: `displaylink-installer.patch'
find: possible unquoted pattern after predicate `-name'?
https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
--2020-11-13 07:54:48--  https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
Resolviendo crazy.dev.frugalware.org (crazy.dev.frugalware.org)... 85.10.194.195
Conectando con crazy.dev.frugalware.org (crazy.dev.frugalware.org)[85.10.194.195]:443... conectado.
Petición HTTP enviada, esperando respuesta... 304 Not Modified
Fichero “evdi-all-in-one-fixes.patch” no modificado en el servidor. Omitiendo la descarga.

Clonando en 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 9 (delta 0), pack-reused 0
Desempaquetando objetos: 100% (46/46), listo.
Nota: actualizando el árbol de trabajo 'dc595db636845aef39490496bc075f6bf067106c'.

Te encuentras en estado 'detached HEAD'. Puedes revisar por aquí, hacer
cambios experimentales y confirmarlos, y puedes descartar cualquier
commit que hayas hecho en este estado sin impactar a tu rama realizando
otro checkout.

Si quieres crear una nueva rama para mantener los commits que has creado,
puedes hacerlo (ahora o después) usando -b con el comando checkout. Ejemplo:

  git checkout -b <nombre-de-nueva-rama>

patching file .travis.yml
Hunk #1 FAILED at 26.
1 out of 1 hunk FAILED -- saving rejects to file .travis.yml.rej
patching file Makefile
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
patching file library/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file module/Makefile
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
patching file module/evdi_drv.c
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
patching file module/evdi_drv.h
Hunk #1 FAILED at 20.
Hunk #2 FAILED at 51.
Hunk #3 FAILED at 112.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
patching file module/evdi_gem.c
Hunk #1 FAILED at 61.
Hunk #2 FAILED at 254.
Hunk #3 FAILED at 569.
Hunk #4 FAILED at 581.
4 out of 4 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module
make[1]: se entra en el directorio '/home/martin/displaylink-debian/evdi/module'
make -C /lib/modules/4.19.0-11-amd64/build M=$PWD
make[2]: se entra en el directorio '/usr/src/linux-headers-4.19.0-11-amd64'
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_drv.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_modeset.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_connector.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_encoder.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_main.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_fb.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_gem.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_painter.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_params.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_cursor.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_debug.o
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_ioc32.o
  LD [M]  /home/martin/displaylink-debian/evdi/module/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/martin/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /home/martin/displaylink-debian/evdi/module/evdi.ko
make[2]: se sale del directorio '/usr/src/linux-headers-4.19.0-11-amd64'
make[1]: se sale del directorio '/home/martin/displaylink-debian/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library
make[1]: se entra en el directorio '/home/martin/displaylink-debian/evdi/library'
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc
cp libevdi.so.1.7.0 libevdi.so
make[1]: se sale del directorio '/home/martin/displaylink-debian/evdi/library'
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux 10 (buster)
Detected running Xorg session and connected docking station
Please disconnect the dock before continuing
Installation terminated.

----- Any lead will be very welcome!

Barabazs, 2020-11-13:
Hi @MartinEtcheverry. I'll close your other, identical, issue and continue here.
```
Detected running Xorg session and connected docking station
Please disconnect the dock before continuing
Installation terminated.
```
According to this message, you are trying to install with the dock connected.

MartinEtcheverry, 2020-11-13:
Ok , i tried to install with the dock disconnected.  but pretty much the same:
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at:
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-11-13 09:38:27--  https://www.displaylink.com/downloads/file?id=1576
Resolviendo www.displaylink.com (www.displaylink.com)... 52.214.244.113, 34.252.69.235, 18.203.121.10, ...
Conectando con www.displaylink.com (www.displaylink.com)[52.214.244.113]:443... conectado.
Petición HTTP enviada, esperando respuesta... 303 See Other
Localización: https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1605271709&Signature=6i1c0eTieZctFzTnNtIyC6ANt%2Bo%3D [siguiendo]
--2020-11-13 09:38:29--  https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1605271709&Signature=6i1c0eTieZctFzTnNtIyC6ANt%2Bo%3D
Resolviendo s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)... 52.218.96.122
Conectando con s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)[52.218.96.122]:443... conectado.
Petición HTTP enviada, esperando respuesta... 200 OK
Longitud: 13889958 (13M) [application/octet-stream]
Grabando a: “DisplayLink_Ubuntu_5.3.1.zip”

DisplayLink_Ubuntu_5.3.1.zip                                    100%[=====================================================================================================================================================>]  13,25M  3,46MB/s    en 4,1s

2020-11-13 09:38:34 (3,24 MB/s) - “DisplayLink_Ubuntu_5.3.1.zip” guardado [13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%

-------------------------------------------------------------------
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias
Leyendo la información de estado... Hecho
libelf-dev ya está en su versión más reciente (0.176-1.1).
libdrm-dev ya está en su versión más reciente (2.4.97-1).
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 1 no actualizados.
find: paths must precede expression: `displaylink-installer.patch'
find: possible unquoted pattern after predicate `-name'?
https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
--2020-11-13 09:38:35--  https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
Resolviendo crazy.dev.frugalware.org (crazy.dev.frugalware.org)... 85.10.194.195
Conectando con crazy.dev.frugalware.org (crazy.dev.frugalware.org)[85.10.194.195]:443... conectado.
Petición HTTP enviada, esperando respuesta... 304 Not Modified
Fichero “evdi-all-in-one-fixes.patch” no modificado en el servidor. Omitiendo la descarga.

[... 59 lines cut ...]
  CC [M]  /home/martin/displaylink-debian/evdi/module/evdi_ioc32.o
  LD [M]  /home/martin/displaylink-debian/evdi/module/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/martin/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /home/martin/displaylink-debian/evdi/module/evdi.ko
make[2]: se sale del directorio '/usr/src/linux-headers-4.19.0-11-amd64'
make[1]: se sale del directorio '/home/martin/displaylink-debian/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library
make[1]: se entra en el directorio '/home/martin/displaylink-debian/evdi/library'
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc
cp libevdi.so.1.7.0 libevdi.so
make[1]: se sale del directorio '/home/martin/displaylink-debian/evdi/library'
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1
DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux 10 (buster)
WARNING: Version 1.6.0 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.

After that rebooted but nothing changed. 
By the way , thanks for the fast reply! :-D

Barabazs, 2020-11-13:
Please post the output of `dkms status`

MartinEtcheverry, 2020-11-13:
oh ok !
root@laptop:/home/martin/displaylink-debian# dkms status
evdi, 1.6.0+dfsg, 4.19.0-11-amd64, x86_64: installed
v4l2loopback, 0.12.1, 4.19.0-11-amd64, x86_64: installed

Barabazs, 2020-11-13:
What does `sudo displaylink-installer` uninstall output?

MartinEtcheverry, 2020-11-13:
sorry for the question , but where is displaylink-installer? i just cloned the repo ... i need to download anything else?

MartinEtcheverry, 2020-11-13:
Sorry , my mistake 

root@laptop:/home/martin/displaylink-debian# /home/martin/displaylink-debian/5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh uninstall
DisplayLink Linux Software 5.3.1.34 install script called: uninstall
Distribution discovered: Debian GNU/Linux 10 (buster)

Uninstalling

[ Removing EVDI from kernel tree, DKMS, and removing sources. ]
dkms remove evdi/1.7.0 --all
Error! There are no instances of module: evdi
1.7.0 located in the DKMS tree.
make: *** [Makefile:88: uninstall_dkms] Error 3
/home/martin/displaylink-debian
Stopping displaylink-driver systemd service
Failed to stop displaylink-driver.service: Unit displaylink-driver.service not loaded.
Failed to disable unit: Unit file displaylink-driver.service does not exist.
[ Removing suspend-resume hooks ]
[ Removing udev rule ]
[ Removing Core folder ]

Uninstallation steps complete.
Please note that the evdi kernel module is still in the memory.
A reboot is required to fully complete the uninstallation process.

Barabazs, 2020-11-13:
No problem at all. The command "displaylink-installer" should be available anywhere if you previously installed (even partially) displaylink. 
Because you already have evdi 1.6.0 installed, it indicates that you there has been a previous install.

Barabazs, 2020-11-13:
After rebooting, please repost the output of `dkms status`

MartinEtcheverry, 2020-11-13:
ok , mhh looks the same ....
root@laptop:/home/martin# dkms status
evdi, 1.6.0+dfsg, 4.19.0-11-amd64, x86_64: installed
v4l2loopback, 0.12.1, 4.19.0-11-amd64, x86_64: installed

Barabazs, 2020-11-13:
Now you should try to uninstall evdi v1.6.0.

1. `sudo dkms remove evdi/1.6.0 --all`
2. Then reboot.
3. And then repost the output of  `dkms status`

MartinEtcheverry, 2020-11-13:
ok , now is clean
root@laptop:/home/martin# dkms status
v4l2loopback, 0.12.1, 4.19.0-11-amd64, x86_64: installed

Barabazs, 2020-11-13:
Now you can try to reinstall displaylink-debian. Please use the updated code from my repo. It should produce a clean install.

1. Delete your current displaylink-debian folder.
2. `git clone https://github.com/Barabazs/displaylink-debian.git && cd displaylink-debian/ && sudo ./displaylink-debian.sh --install`
3. Copy the output of the installation and save it somewhere.
4. Reboot and see if it works.
5. Repost the output of the installation here.

MartinEtcheverry, 2020-11-13:
git clone https://github.com/Barabazs/displaylink-debian.git && cd displaylink-debian/ && sudo ./displaylink-debian.sh --install
Clonando en 'displaylink-debian'...
remote: Enumerating objects: 55, done.
remote: Counting objects: 100% (55/55), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 945 (delta 32), reused 20 (delta 10), pack-reused 890
Recibiendo objetos: 100% (945/945), 228.21 KiB | 359.00 KiB/s, listo.
Resolviendo deltas: 100% (413/413), listo.

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at:
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-11-13 11:18:43--  https://www.displaylink.com/downloads/file?id=1576
Resolviendo www.displaylink.com (www.displaylink.com)... 18.203.121.10, 52.214.244.113, 34.252.69.235, ...
Conectando con www.displaylink.com (www.displaylink.com)[18.203.121.10]:443... conectado.
Petición HTTP enviada, esperando respuesta... 303 See Other
Localización: https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1605277724&Signature=teWGxGx%2BNLmCRIT6U7AyQGcDBd8%3D [siguiendo]
--2020-11-13 11:18:44--  https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1605277724&Signature=teWGxGx%2BNLmCRIT6U7AyQGcDBd8%3D
Resolviendo s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)... 52.218.62.107
Conectando con s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)[52.218.62.107]:443... conectado.
Petición HTTP enviada, esperando respuesta... 200 OK
Longitud: 13889958 (13M) [application/octet-stream]
Grabando a: “DisplayLink_Ubuntu_5.3.1.zip”
DisplayLink_Ubuntu_5.3.1.zip 100%[===========================================>]  13,25M   565KB/s    en 33s

2020-11-13 11:19:18 (412 KB/s) - “DisplayLink_Ubuntu_5.3.1.zip” guardado [13889958/13889958]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%

-------------------------------------------------------------------
Clonando en 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 9 (delta 0), pack-reused 0
Desempaquetando objetos: 100% (46/46), listo.
Nota: actualizando el árbol de trabajo 'dc595db636845aef39490496bc075f6bf067106c'.

Te encuentras en estado 'detached HEAD'. Puedes revisar por aquí, hacer
[... 96 lines cut ...]
Distribution discovered: Debian GNU/Linux 10 (buster)
WARNING: Version 1.7.0 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.ylink-debian/evdi/module/evdi.ko
make[2]: se sale del directorio '/usr/src/linux-headers-4.19.0-11-amd64'
make[1]: se sale del directorio '/home/martin/displaylink-debian/displaylink-debian/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library
make[1]: se entra en el directorio '/home/martin/displaylink-debian/displaylink-debian/evdi/library'
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc
cp libevdi.so.1.7.0 libevdi.so
make[1]: se sale del directorio '/home/martin/displaylink-debian/displaylink-debian/evdi/library'
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux 10 (buster)
WARNING: Version 1.7.0 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.

After that restarted , but i check the dkms status and find that:
root@laptop:/home/martin# dkms status
v4l2loopback, 0.12.1, 4.19.0-11-amd64, x86_64: installed

Barabazs, 2020-11-13:
Strange...
Please post the output of `sudo ./displaylink-debian.sh --debug`

MartinEtcheverry, 2020-11-13:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
DisplayLink service status:
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Core
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D):
X11 version: 1.20.4-1+deb10u1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No existe el fichero o el directorio
Contents:

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2020-11-13:
Can you confirm that dkms status still does not reference evdi?

MartinEtcheverry, 2020-11-13:
yes

MartinEtcheverry, 2020-11-13:
root@laptop:/home/martin/displaylink-debian# dkms status
v4l2loopback, 0.12.1, 4.19.0-11-amd64, x86_64: installed
root@laptop:/home/martin/displaylink-debian# dkms status
v4l2loopback, 0.12.1, 4.19.0-11-amd64, x86_64: installed

Barabazs, 2020-11-13:
Can you provide output of `sudo dkms status evdi`

MartinEtcheverry, 2020-11-13:
ok , i managed to get the service installed 

apt remove evdi-dkms
cd displaylink-debian
rmmod evdi
./displaylink-debian.sh

then installed perfectly ...and ask for reboot. 
now 
root@laptop:/home/martin# xrandr --listproviders
Providers: number : 5
Provider 0: id: 0xce cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

Barabazs, 2020-11-13:
Good to hear! Now you need to link your providers [1-4] to provider 0.
You can find detailed instructions here: https://github.com/Barabazs/displaylink-debian/blob/master/post-install-guide.md#display-detection

MartinEtcheverry, 2020-11-13:
Wow! excelent!
Thank you so much for your time and patience!

Barabazs, 2020-11-13:
You're welcome. I'm glad it worked out.

closed 2020-11-13 by Barabazs