AdnanHodzic/displaylink-debian #502: Display Asus Zenscreen MB16AC not detected with my Debian Buster stable version

HoangTienTranNgoc, 2020-11-14:
Hello Adnan,

I have an issue with my Asus zenscreen. Before i had an windows 10 OS installed on my Asus Rog G75V, it worked well. Now i install a debian buster stable version, my zenscreen is no more detected. I use an usb-c /usb-3 adapter. I have read and try all your advices and made research on the web and can't fix it. 
I have a second question  how to save the xrandr --setprovideroutputsource 1 0 after reboot.
So i will ask for your expertise and the time you will pass to help me.
I will first thank you for your help and answer.
So i'll give you all the information i have:
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-12-amd64

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: GK104M
VGA: NVIDIA Corporation GK104M [GeForce GTX 670MX] (rev a1)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x301 cap: 0x1, Source Output crtcs: 4 outputs: 5 associated providers: 4 name:NVIDIA-0
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

displaylink-debian# lsusb
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 1bcf:2885 Sunplus Innovation Technology Inc. ASUS Webcam
Bus 001 Device 003: ID 13d3:3404 IMC Networks 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0bda:0412 Realtek Semiconductor Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 004: ID 0bda:5400 Realtek Semiconductor Corp. 
Bus 002 Device 003: ID 0bda:5412 Realtek Semiconductor Corp. 
Bus 002 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

xrandr
Screen 0: minimum 8 x 8, current 1920 x 1080, maximum 16384 x 16384
VGA-0 disconnected (normal left inverted right x axis y axis)
LVDS-0 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 381mm x 214mm
   1920x1080     60.04*+
DP-0 disconnected (normal left inverted right x axis y axis)
HDMI-0 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
DVI-I-1-1 disconnected (normal left inverted right x axis y axis)

Barabazs, 2020-11-14:
Hi @HoangTienTranNgoc. 

1. As far as I know, the setprovideroutputsource command should only be set once and is persistent across reboots. 
2. Is all the content in 20-displaylink.conf installed by default or did you add any of the configs yourself?

HoangTienTranNgoc, 2020-11-14:
Hello Barabazs 
First, thanks to answer me.
i think it was the default but i tried so many manipulations to make it functionnal that i m not sure...
Plus, i had to reinstall my whole system, since my first post. I have to follow the Adnan's advices once. Or may be i can follow your advices to install the zenscreen step by step.

Barabazs, 2020-11-14:
In that case you should start by reinstalling displaylink-debian, copy the install output here.
Then run the debug procedure and post the output here.

HoangTienTranNgoc, 2020-11-14:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] Y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] Y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-12-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: GK104M
VGA: NVIDIA Corporation GK104M [GeForce GTX 670MX] (rev a1)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x301 cap: 0x1, Source Output crtcs: 4 outputs: 5 associated providers: 0 name:NVIDIA-0
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
I don't have the install log because i have restarted the computer, i post you the debug info. I do not make the provider setting yet. i follow your advice.

Barabazs, 2020-11-14:
Can you try the providersetting procedure now please.

HoangTienTranNgoc, 2020-11-14:
Done.
xrandr --listproviders
Providers: number : 5
Provider 0: id: 0x301 cap: 0x1, Source Output crtcs: 4 outputs: 5 associated providers: 4 name:NVIDIA-0
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

HoangTienTranNgoc, 2020-11-14:
I still not connect my zenscreen until now.

Barabazs, 2020-11-14:
> I still not connect my zenscreen until now.

Do you mean it is not plugged in?

HoangTienTranNgoc, 2020-11-14:
yes

Barabazs, 2020-11-14:
Then you should plug it in.

HoangTienTranNgoc, 2020-11-14:
May be i have to reboot after the provider setup before?

HoangTienTranNgoc, 2020-11-14:
I reboot and plug my zenscreen but it's always the same message: Your device support not the ALT-DP mode.
I think the charge is ok because the zenscreen power is on, with the Asus logo and react.

Barabazs, 2020-11-14:
You didn't mention that error message the first time...

You should force the screen to use usb mode instead of ALT-DP mode.
Since displaylink-debian correctly installed evdi and displaylink, the configs looks ok and the providers are set, the issue is unrelated to the use of displaylink-debian.

HoangTienTranNgoc, 2020-11-14:
I've always seen this message even on my windows 10 installation. When the laptop wasn't powered. I thought it was because the correct driver on my debian OS was not well installed.
i will try to force it but i don't know how. The two buttons on the left of the screen don't react. i go try on a windows computer.

HoangTienTranNgoc, 2020-11-14:
i check on a windows 10 computer, my screen connect by usb without problems.
I have to force displaylink to connect by usb, there no options on my zenscreen to force the usb connection.

HoangTienTranNgoc, 2020-11-14:
I found this :Bus=03 Lev=02 Prnt=08 Port=01 Cnt=01 Dev#=  9 Spd=5000 MxCh= 0
D:  Ver= 3.20 Cls=ef(misc ) Sub=02 Prot=01 MxPS= 9 #Cfgs=  1
P:  Vendor=17e9 ProdID=4374 Rev=31.01
S:  Manufacturer=DisplayLink
S:  Product=MB16AC
S:  SerialNumber=4866484M0503
C:* #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=504mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=03 Driver=usbfs
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=84(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=usbfs

HoangTienTranNgoc, 2020-11-14:
And this by a lsusb command:
lsusb
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 1bcf:2885 Sunplus Innovation Technology Inc. ASUS Webcam
Bus 001 Device 003: ID 13d3:3404 IMC Networks 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0bc2:231a Seagate RSS LLC Expansion Portable
Bus 003 Device 009: ID 17e9:4374 DisplayLink 
Bus 003 Device 008: ID 0bda:0412 Realtek Semiconductor Corp. 
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 014: ID 0bda:5400 Realtek Semiconductor Corp. 
Bus 002 Device 013: ID 0bda:5412 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Barabazs, 2020-11-14:
You'll have to contact Asus support or try to find someone with a similar setup. 

IIRC, to force your screen in USB mode, you have to connect it to a USB-A port, but then you have to find another way to power your screen. But even then, there is no guarantee that it will work...

closed 2020-11-14 by Barabazs

HoangTienTranNgoc, 2020-11-14:
My screen is already in usb mode when i connect to another computer with windows 10, a message on the screen inform us that the mode is USB. No way to ofrec it from the screen. And the screen is powered cause when i plug it in the USB-A port of the computer the screen power button is on and we can see the asus logo. The problem doesn't come frome the screen mode or the power. You have right when you say that the displaylink recognize the screen we can see it in the lsusb report. The cause is surely in another way.

HoangTienTranNgoc, 2020-11-14:
I thank you for the time you take to help me, anyways.

Barabazs, 2020-12-14:
@HoangTienTranNgoc FYI Someone with the same screen as yours managed to make it work. Have a look at #513