AdnanHodzic/displaylink-debian #504: The second Monitor works slower than the first

schmichris2, 2020-11-18:
Hi,
I have a problem with my second Monitor he works very slow (Mouse, Video etc). Have you any idea whats the problem is? Thank You! Hear are the Output: 

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-54-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Ridge
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c4)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.4

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Datei oder Verzeichnis nicht gefunden
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x55 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:05:00.0
Provider 1: id: 0x108 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xe6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xc4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xa0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Barabazs, 2020-11-18:
When is it slow? Is your laptop monitor on or off?

schmichris2, 2020-11-18:
Thank you!

> When is it slow? 
All time!
For example: If I look a video, it looks like missing frames an the video is not sync with the audio.
Mouse-Pointer: Moving the mouse pointer is much slower than with the other monitor.

Is your laptop monitor on or off?
I've tried both variants, the problem remains even when the laptop monitor is on or off.

I previously installed the official driver (displaylink-driver-5.3.1.34.run) that was included in the documentation for the USB graphics card. Is this a problem? 
In the documentation there was a note if there were problems with the official driver (e.g. slowly) you should try this driver version.

Barabazs, 2020-11-18:
> I previously installed the official driver (displaylink-driver-5.3.1.34.run) that was included in the documentation for the USB graphics card. Is this a problem?
> In the documentation there was a note if there were problems with the official driver (e.g. slowly) you should try this driver version.

This is unrelated to displaylink-debian. You should try to get support from DisplayLink on their forum.

closed 2020-11-18 by Barabazs