AdnanHodzic/displaylink-debian #505: No version in info - monitors not working

agnesdennetiere, 2020-11-23:
Hello,

I installed this (advice from the manufacturer of my docking station since it was slow)
but when I enter the command `sudo ./displaylink-debian.sh --debug` , I get the following (no driver version) and my monitors are not detected anymore.

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-54-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Ltd.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c7)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.4

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x54 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 1 associated providers: 1 name:Unknown AMD Radeon GPU @ pci:0000:07:00.0
Provider 1: id: 0x84 cap: 0x6, Sink Output, Source Offload crtcs: 5 outputs: 1 associated providers: 1 name:Radeon RX 5500M @ pci:0000:03:00.0
```

I tried to reinstall but I get this message:

```
Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: Aucun fichier ou dossier de ce type
```

What can I do?

Thank you

Barabazs, 2020-11-23:
Hi @agnesdennetiere. It looks like you didn't install it properly. Please try again and copy the output here. Don't forget to install with the dock disconnected.

Can you share the communication you had with the manufacturer please?

Barabazs, 2020-11-28:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2020-11-28 by Barabazs

agnesdennetiere, 2020-12-01:
Hi there,

sorry for the late answer.
I installed the driver again and it worked, but it was still slow, so I used amdgpu pro and it seems to work for me (https://www.amd.com/en/support/kb/release-notes/rn-amdgpu-unified-linux-20-20)

Thank you!

Barabazs, 2020-12-01:
Glad it worked out.