AdnanHodzic/displaylink-debian #508: LMDE 4 not able to reboot after displaylink-debian installation

vama22, 2020-12-02:
Hi,

After installation of displaylink-debian on LMDE 4 full updated, it is not able to reboot : Bluetooth hci0 Failed to send firmware data.

I've disabled Secure Boot and I tried multiple times.

Thanks a lot for your helping.

Regards.

Barabazs, 2020-12-02:
Please provide more info and logs.

vama22, 2020-12-03:
I've just launched the command "sudo ./displaylink-debian.sh", the installation completes and it asks to reboot. After that, the system is not able to boot with errors in attachment.
[image]

I am at disposal to give you information you need.

Barabazs, 2020-12-03:
Please provide the logs for the period when you installed with displaylink-debian and logs when booting/rebooting.
I'm not sure why this would be related to the use of displaylink-debian...

The error seems pretty common and there is a lot of info online that might help you solve this issue.

vama22, 2020-12-04:
I installed 4 times LDME and it is not able to reboot just after installing displaylink-debian. Also, before the first displaylink-debian installation, my LDME was working right since several months.

Due to the incapacity to reboot the system after installation, which logs do you want and from where?

Barabazs, 2020-12-04:
You'll have to chroot from a liveCD and uninstall displaylink-debian. But I highly doubt that it will fix your issue.

vama22, 2020-12-04:
I'm sorry, but I don't thing that is out of scope, because I use a i-Tec docking station and they say to install displaylink-debian. I need i-tec drivers to use my screens and after installation I'm not able to reboot. Perhaps this is information could be helpful.

Have you tried to install again on LDME 4?

Thanks a lot.

closed 2020-12-14 by Barabazs