AdnanHodzic/displaylink-debian #509: Installer on Debian is not working

svn-webmaster, 2020-12-04:
Hi,

on my debian 10.6 with 5.8 Kernel i got the following message by trying to run the installer:

`root@laptop-hp:~/displaylink-debian# ./displaylink-debian.sh
Can't open display `

That's all - no further informations.

My DisplayLink device :
`root@laptop-hp:~/displaylink-debian# lsusb -d 17e9:
Bus 001 Device 010: ID 17e9:0290 DisplayLink `
`ec  4 23:49:08 laptop-hp kernel: [16495.012102] usb 1-2.3: new high-speed USB device number 10 using xhci_hcd
Dec  4 23:49:13 laptop-hp kernel: [16500.224542] usb 1-2.3: device descriptor read/64, error -110
Dec  4 23:49:14 laptop-hp kernel: [16500.482635] usb 1-2.3: New USB device found, idVendor=17e9, idProduct=0290, bcdDevice= 0.01
Dec  4 23:49:14 laptop-hp kernel: [16500.482638] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec  4 23:49:14 laptop-hp kernel: [16500.482641] usb 1-2.3: Product: DL-165 Adapter
Dec  4 23:49:14 laptop-hp kernel: [16500.482642] usb 1-2.3: Manufacturer: DisplayLink
Dec  4 23:49:14 laptop-hp kernel: [16500.482644] usb 1-2.3: SerialNumber: 124300
Dec  4 23:49:14 laptop-hp kernel: [16500.526841] udlfb 1-2.3:1.0: vendor descriptor length: 34 data: 22 5f 01 00 20 05 00 01 03 04 02
Dec  4 23:49:14 laptop-hp kernel: [16500.526844] udlfb 1-2.3:1.0: DL chip limited to -17152 pixel modes
Dec  4 23:49:14 laptop-hp kernel: [16500.629994] usb 1-2.3: fb1 is DisplayLink USB device (1280x1024, 5120K framebuffer memory)`

Kernel-Sources and headers are installed. 

Are more information's needed? 

Thanks for your help!

Barabazs, 2020-12-05:
Can you post the output of `sudo ./displaylink-debian.sh --debug` please.

svn-webmaster, 2020-12-05:
> 

> Can you post the output of `sudo ./displaylink-debian.sh --debug` please.

same result:
`root@laptop-hp:~/displaylink-debian# ./displaylink-debian.sh --debug
Can't open display `

Barabazs, 2020-12-05:
And the output of:
```
lsb_release -a
uname --all
xrandr --listproviders
```

svn-webmaster, 2020-12-05:
lsb_release -a
`root@laptop-hp:~/displaylink-debian# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
`
uname --all
`Linux laptop-hp 5.8.0-0.bpo.2-amd64 #1 SMP Debian 5.8.10-1~bpo10+1 (2020-09-26) x86_64 GNU/Linux`

xrandr --listproviders
`root@laptop-hp:~/displaylink-debian# xrandr --listproviders
Can't open display `

Barabazs, 2020-12-05:
Why are you logged in as root?

svn-webmaster, 2020-12-05:
I'm logged in as normal user, but in my terminal-emulator I'm using the su command to save the sudo command. 

as normal user and sudo the script works! 
Now I have other issues but I think the Post Installation Guide should help to solve them.

Thanks

Barabazs, 2020-12-05:
OK, good to hear it's working now.

closed 2020-12-05 by Barabazs