AdnanHodzic/displaylink-debian #511: Parrot Os kernel 5.9, asus mb169b

n1f1n, 2020-12-05:
Hello.
After upgraded my parrot, my monitor stoped working :(
Details:

#uname -a
Linux parrot 5.9.0-2parrot1-amd64 #1 SMP Debian 5.9.6-2parrot1 (2020-11-17) x86_64 GNU/Linux

#cd displaylink-debian/ && sudo ./displaylink-debian.sh

--------------------------- displaylink-debian -------------------------------                                                                                                                                 
                                                                                                                                                                                                               
DisplayLink driver installer for Debian and Ubuntu based Linux distributions:                                                                                                                                  
                                                                                                                                                                                                               
* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!                                                                                                                                             
* Full list of all supported platforms: http://bit.ly/2zrwz2u                                                                                                                                                  
* When submitting a new issue, include Debug information                                                                                                                                                       
                                                                                                                                                                                                               
Options:                                                                                                                                                                                                       
                                                                                                                                                                                                               
[I]nstall                                                                                                                                                                                                      
[D]ebug                                                                                                                                                                                                        
[R]e-install                                                                                                                                                                                                   
[U]ninstall                                                                                                                                                                                                    
[Q]uit                                                                                                                                                                                                         
                                                                                                                                                                                                               
Select a key: [i/d/r/u/q]: i                                                                                                                                                                                   
                                                                                                                                                                                                               
-------------------------------------------------------------------                                                                                                                                            
                                                                                                                                                                                                               
Checking dependencies                                                                                                                                                                                          
                                                                                                                                                                                                               
unzip is installed                                                                                                                                                                                             
linux-headers-5.9.0-2parrot1-amd64 is installed                                                                                                                                                                
dkms is installed                                                                                                                                                                                              
lsb-release is installed                                                                                                                                                                                       
linux-source is installed                                                                                                                                                                                      
x11-xserver-utils is installed                                                                                                                                                                                 
wget is installed                                                                                                                                                                                              
libdrm-dev is installed                                                                                                                                                                                        
git is installed                                                                                                                                                                                               
                                                                                                                                                                                                               
Platform requirements satisfied, proceeding ...                                                                                                                                                                
                                                                                                                                                                                                               
-------------------------------------------------------------------                                                                                                                                            
                                                                                                                                                                                                               
Please read the Software License Agreement available at:                                                                                                                                                       
https://www.displaylink.com/downloads/file?id=1576                                                                                                                                                             
Do you accept?: [Y/n]: y                                                                                                                                                                                       
[... cut ...]
make[3]: *** [/usr/src/linux-headers-5.9.0-2parrot1-common/Makefile:1796: /home/z1f1z/Downloads/Sorted-downloads/displaylink-debian/evdi/module] Error 2
make[2]: *** [/usr/src/linux-headers-5.9.0-2parrot1-common/Makefile:185: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.9.0-2parrot1-amd64'
make[1]: *** [Makefile:71: module] Error 2
make[1]: Leaving directory '/home/z1f1z/Downloads/Sorted-downloads/displaylink-debian/evdi/module'
make: *** [Makefile:8: all] Error 2
cp: cannot stat 'library/libevdi.so.1.*': No such file or directory

Thanks in advance.

Barabazs, 2020-12-05:
You're using an older version of displaylink-debian. Uninstall first ans then pull the latest from git.

n1f1n, 2020-12-05:
ok, i did it.
result:

--------------- Linux system info ----------------                                                                                                                                                             
                                                                                                                                                                                                               
Distro: Parrot                                                                                                                                                                                                 
Release: n/a                                                                                                                                                                                                   
Kernel: 5.9.0-2parrot1-amd64                                                                                                                                                                                   
                                                                                                                                                                                                               
---------------- DisplayLink info ----------------                                                                                                                                                             
                                                                                                                                                                                                               
Driver version: 1.7.0                                                                                                                                                                                          
DisplayLink service status:                                                                                                                                                                                    
EVDI service version: 1.7.0                                                                                                                                                                                    
                                                                                                                                                                                                               
------------------ Graphics card -----------------                                                                                                                                                             
                                                                                                                                                                                                               
Vendor: i915                                                                                                                                                                                                   
Subsystem: UHD                                                                                                                                                                                                 
VGA: Intel Corporation UHD Graphics 630 (Mobile)                                                                                                                                                               
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)                                                                                                                                      
X11 version: 1.20.8-2                                                                                                                                                                                          
X11 configs: /etc/X11/xorg.conf.d/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf                                                                                                                           
                                                                                                                                                                                                               
-------------- DisplayLink xorg.conf -------------                                                                                                                                                             

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "intel"
    Screen 1 "nvidia"
    #Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    #Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x2, Sink Output crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x16e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x14c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x12a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x108 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

n1f1n, 2020-12-05:
### UPDATE

**Thanks** working !!!

Barabazs, 2020-12-06:
Great!
Can you post the output of `sudo ./displaylink-debian.sh --debug` for future reference please?

closed 2020-12-06 by Barabazs

n1f1n, 2020-12-06:
Sure (temészetesen) !

-------------------------------------------------------------------                                                                                                                                            
                                                                                                                                                                                                               
Starting Debug ...                                                                                                                                                                                             
                                                                                                                                                                                                               
Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y                                                                                                                                            
                                                                                                                                                                                                               
Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y                                                                                                                                 
                                                                                                                                                                                                               
--------------- Linux system info ----------------                                                                                                                                                             
                                                                                                                                                                                                               
Distro: Parrot                                                                                                                                                                                                 
Release: n/a                                                                                                                                                                                                   
Kernel: 5.9.0-2parrot1-amd64                                                                                                                                                                                   
                                                                                                                                                                                                               
---------------- DisplayLink info ----------------                                                                                                                                                             
                                                                                                                                                                                                               
Driver version: 1.7.0                                                                                                                                                                                          
DisplayLink service status: up and running                                                                                                                                                                     
EVDI service version: 1.7.0                                                                                                                                                                                    
                                                                                                                                                                                                               
------------------ Graphics card -----------------                                                                                                                                                             
                                                                                                                                                                                                               
Vendor: i915                                                                                                                                                                                                   
Subsystem: UHD                                                                                                                                                                                                 
VGA: Intel Corporation UHD Graphics 630 (Mobile)                                                                                                                                                               
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)                                                                                                                                      
[... cut ...]
Provider 2: id: 0x15c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------