AdnanHodzic/displaylink-debian #512: Ubuntu 20.10

FandaOG, 2020-12-07:
Hi, I tried to install drivers on Ubuntu 20.10 (Kubuntu 20.10) but I've got an error:
```
make[2]: Entering directory '/usr/src/linux-headers-5.8.0-31-generic'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.
```
I have linux headers installed. I've tried to install https://launchpad.net/ubuntu/groovy/+package/libevdi0 but it does not help. Do you know how to fix it? Thanks.

Full log:
```
Downloading DisplayLink Ubuntu driver:

--2020-12-07 08:50:38--  https://www.displaylink.com/downloads/file?id=1576
Resolving www.displaylink.com (www.displaylink.com)... 52.214.244.113, 34.252.69.235, 34.250.82.133, ...
Connecting to www.displaylink.com (www.displaylink.com)|52.214.244.113|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1607328039&Signature=MSM%2Fc40%2F9VwrmEUtCBAHKdTKGnQ%3D [following]
--2020-12-07 08:50:39--  https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1607328039&Signature=MSM%2Fc40%2F9VwrmEUtCBAHKdTKGnQ%3D
Resolving s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)... 52.218.90.19
Connecting to s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)|52.218.90.19|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13889958 (13M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_5.3.1.zip’

DisplayLink_Ubuntu_5.3.1.zip           100%[==========================================================================>]  13,25M  1,69MB/s    in 7,5s    

2020-12-07 08:50:46 (1,76 MB/s) - ‘DisplayLink_Ubuntu_5.3.1.zip’ saved [13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  

-------------------------------------------------------------------
Cloning into 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 9 (delta 0), pack-reused 0
Unpacking objects: 100% (46/46), 67.45 KiB | 211.00 KiB/s, done.
Note: switching to '85f3ec152ac67a227123e0a383a091f9c340d796'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

cat: /etc/redhat-release: No such file or directory
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1]: Entering directory '/home/mazura/Documents/smazat/displaylink-debian/evdi/module'
make -C /lib/modules/5.8.0-31-generic/build M=$PWD
make[2]: Entering directory '/usr/src/linux-headers-5.8.0-31-generic'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[2]: *** [Makefile:746: include/config/auto.conf] Error 1
make[2]: Leaving directory '/usr/src/linux-headers-5.8.0-31-generic'
make[1]: *** [Makefile:71: module] Error 2
make[1]: Leaving directory '/home/mazura/Documents/smazat/displaylink-debian/evdi/module'
make: *** [Makefile:13: all] Error 2
cp: cannot stat 'library/libevdi.so.1.*': No such file or directory
```

Barabazs, 2020-12-07:
Please post the output of `sudo ./displaylink-debian.sh --debug`

FandaOG, 2020-12-07:
I tried to install driver from DisplayLink page manually. Output is:

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: groovy
Kernel: 5.8.0-31-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.7.0+dfsg
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
Provider 1: id: 0xef cap: 0x6, Sink Output, Source Offload crtcs: 5 outputs: 0 associated providers: 1 name:Radeon 500 Series @ pci:0000:03:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Thanks.

Barabazs, 2020-12-07:
Your issue seems unrelated to displaylink-debian. The error indicates there might be something wrong with your kernel or a bug in evdi.

Maybe try  to up/downgrade your kernel and try again. Or file an issue upstream.

I'll leave this open for now.

FandaOG, 2020-12-07:
The problem was in kernel. I downgrade kernel to 5.7.19 and it wokrs. Thank you.

closed 2020-12-07 by Barabazs