AdnanHodzic/displaylink-debian #513: LinuxMint 20 on Asus Vivobook will not boot after installing displaylink driver

nd4c, 2020-12-13:
These attempts have been on relatively fresh installs of Linux Mint 20, with a just the initial recommended updates.
The first time I tried, I used the script from the DisplayLink site.  The system would not boot.  I had to reinstall LM 20...
This second time I used TimeShift before trying.  And I followed the instructions from: https://github.com/AdnanHodzic/displaylink-debian.
On reboot, after selecting LM20 from the boot menu, the system just hangs there with a cursor on a black screen.
So I had to use TimeShift to get back up and running.
I ran "sudo ./displaylink-debian.sh --debug" even though the driver is not installed, in hopes it might give you some other info:

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-58-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: 
Subsystem: Renoir
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c2)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.6

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2020-12-13:
Hi @nd4c. I will need more info to be able to look into this issue.
I will need:
1.  the terminal output when installing with displaylink-debian
2. logs/error messages/... when trying to reboot and the boot fails

nd4c, 2020-12-13:
Output from sudo ./displaylink-debian.sh

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.4.0-58-generic is installed
dkms is installed
lsb-release is installed
dpkg-query: package 'linux-source' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
linux-source not found! Install? [Y/n] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  linux-source-5.4.0
Suggested packages:
  libncurses-dev | ncurses-dev kernel-package libqt3-dev
The following NEW packages will be installed:
  linux-source linux-source-5.4.0
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/135 MB of archives.
After this operation, 150 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package linux-source-5.4.0.
(Reading database ... 332060 files and directories currently installed.)
Preparing to unpack .../linux-source-5.4.0_5.4.0-58.64_all.deb ...
Unpacking linux-source-5.4.0 (5.4.0-58.64) ...
Selecting previously unselected package linux-source.
Preparing to unpack .../linux-source_5.4.0.58.61_all.deb ...
Unpacking linux-source (5.4.0.58.61) ...
Setting up linux-source-5.4.0 (5.4.0-58.64) ...
Setting up linux-source (5.4.0.58.61) ...
x11-xserver-utils is installed
wget is installed
dpkg-query: package 'libdrm-dev' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
libdrm-dev not found! Install? [Y/n] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libdrm-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/126 kB of archives.
After this operation, 658 kB of additional disk space will be used.
Selecting previously unselected package libdrm-dev:amd64.
(Reading database ... 332542 files and directories currently installed.)
Preparing to unpack .../libdrm-dev_2.4.101-2_amd64.deb ...
Unpacking libdrm-dev:amd64 (2.4.101-2) ...
[... 204 lines cut ...]

-------------------------------------------------------------------

Performing post install steps

Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Setup DisplayLink xorg.conf depending on graphics card
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.3.1.zip" file
Removing redundant: "5.3.1" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes
After reboot, make sure to consult post-install guide! http://bit.ly/2TbZleK
Reboot now? [Y/n] n

Reboot postponed, changes won't be applied until reboot

-------------------------------------------------------------------

nd4c, 2020-12-13:
Not sure what logs you need.

This the Xorg.0.log.old file:

[     6.109] 
X.Org X Server 1.20.8
X Protocol Version 11, Revision 0
[     6.109] Build Operating System: Linux 4.15.0-124-generic x86_64 Ubuntu
[     6.109] Current Operating System: Linux mintygreen 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64
[     6.109] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-58-generic root=UUID=c72ba3ba-6f52-4085-9bb9-986b86bca15d ro quiet splash
[     6.109] Build Date: 30 November 2020  05:56:33PM
[     6.109] xorg-server 2:1.20.8-2ubuntu2.6 (For technical support please see http://www.ubuntu.com/support) 
[     6.109] Current version of pixman: 0.38.4
[     6.109] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[     6.109] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     6.109] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Dec 13 01:05:17 2020
[     6.109] (==) Using config directory: "/etc/X11/xorg.conf.d"
[     6.109] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     6.109] (==) No Layout section.  Using the first Screen section.
[     6.109] (==) No screen section available. Using defaults.
[     6.109] (**) |-->Screen "Default Screen Section" (0)
[     6.109] (**) |   |-->Monitor "<default monitor>"
[     6.109] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[     6.109] (==) Automatically adding devices
[     6.109] (==) Automatically enabling devices
[     6.109] (==) Automatically adding GPU devices
[     6.109] (==) Automatically binding GPU devices
[     6.110] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     6.110] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     6.110] 	Entry deleted from font path.
[     6.110] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     6.110] 	Entry deleted from font path.
[     6.110] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     6.110] 	Entry deleted from font path.
[     6.110] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[     6.110] 	Entry deleted from font path.
[     6.110] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[     6.110] 	Entry deleted from font path.
[     6.110] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/Type1,
	built-ins
[     6.110] (==) ModulePath set to "/usr/lib/xorg/modules"
[     6.110] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[     6.110] (II) Loader magic: 0x55844137a020
[     6.110] (II) Module ABI versions:
[     6.110] 	X.Org ANSI C Emulation: 0.4
[     6.110] 	X.Org Video Driver: 24.1
[     6.110] 	X.Org XInput driver : 24.1
[     6.110] 	X.Org Server Extension : 10.0
[     6.111] (++) using VT number 7

[     6.111] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[     6.111] (II) xfree86: Adding drm device (/dev/dri/card0)
[     6.112] (II) xfree86: Adding drm device (/dev/dri/card1)
[     6.112] (II) xfree86: Adding drm device (/dev/dri/card2)
[     6.112] (II) xfree86: Adding drm device (/dev/dri/card3)
[     6.115] (--) PCI:*(3@0:0:0) 1002:1636:1043:1352 rev 194, Mem @ 0xd0000000/268435456, 0xe0000000/2097152, 0xfe700000/524288, I/O @ 0x0000f000/256
[     6.115] (II) LoadModule: "glx"
[     6.115] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     6.116] (II) Module glx: vendor="X.Org Foundation"
[     6.116] 	compiled for 1.20.8, module version = 1.0.0
[     6.116] 	ABI class: X.Org Server Extension, version 10.0
[     6.116] (II) Applying OutputClass "DisplayLink" to /dev/dri/card0
[     6.116] 	loading driver: modesetting
[... 169 lines cut ...]
[     6.173] (II) Loading sub module "fbdevhw"
[     6.173] (II) LoadModule: "fbdevhw"
[     6.173] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[     6.173] (II) Module fbdevhw: vendor="X.Org Foundation"
[     6.173] 	compiled for 1.20.8, module version = 0.0.2
[     6.173] 	ABI class: X.Org Video Driver, version 24.1
[     6.173] (**) FBDEV(2): claimed PCI slot 3@0:0:0
[     6.173] (II) FBDEV(2): using default device
[     6.173] (II) modeset(G0): using drv /dev/dri/card0
[     6.173] (II) modeset(G1): using drv /dev/dri/card1
[     6.173] (II) modeset(G2): using drv /dev/dri/card2
[     6.173] (II) modeset(G3): using drv /dev/dri/card3
[     6.173] (EE) Screen 0 deleted because of no matching config section.
[     6.173] (II) UnloadModule: "modesetting"
[     6.173] (EE) 
Fatal server error:
[     6.173] (EE) Cannot run in framebuffer mode. Please specify busIDs        for all framebuffer devices
[     6.173] (EE) 
[     6.174] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[     6.174] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[     6.174] (EE) 
[     6.180] (EE) Server terminated with error (1). Closing log file.

nd4c, 2020-12-13:
Just a note that might be important.  This is a brand new Asus Vivobook and I think the video hardware is new.  It came with win10 - which has no problem working with my Asus portable monitor that requires USB-c Display Port Mode.  However, LinuxMint doesn't seem to have proper display drivers for the hardware.  On boot it warns me that the graphics display is not using hardware excelaration and to look for device drivers.  Linux Mint Device Manager says I don't need any drivers.  

My first attempt at installing Linux Mint, the live iso couldn't display properly.  This newer iso apparently can, but is still using drivers without hardware exceleration.  I've also tried Fedora 33, the main display worked fine, but I could only find displaylink drivers for up to Fedora 32.   Maybe I'll try that next.

Barabazs, 2020-12-13:
When you talk about the black screen, are you talking about the built-in display from your laptop or the external monitor?

nd4c, 2020-12-13:
I'm talking about the built-in display.  The external monitor is an ASUS MB16AC and does not work off the bios.  I've tried the install with and without it plugged into the USB-c connector.  Again, this works in Windows 10, so it's not (exactly) bad hardware.  The external display after attempting to power on (whether connected while in bios or linux) will display a message that the computer does not support USB-c DP Mode.

My current guess is that the DisplayLink drivers expect the graphic drivers to be at a certain level, but the new laptop hardware is not fully supported yet.  But that's just a guess.

The laptop is an ASUS Vivobook 15 S513 with AMD Ryzen 7 4700U.  Apparently the GPU part is called Renoir (rev c2).

The first time I ran a live LM20, the display ran as only VGA, it couldn't support the hardware.  So I tried Fedora 33, and that worked, at least, apparently.  However, the DisplayLink driver install failed in the same way, a cursor on a blank screen.  So I downloaded the latest LM20 and the live handled the display better this time, but after install, every time it boots up, there's a message about the graphics drivers.

If you can figure this out, I would greatly appreciate it, but I'm beginning to suspect that's it the graphics drivers, not the displaylink drivers.  So don't work too hard on it, please.  I might just have to wait a little bit for the AMD drivers to catch up.

Barabazs, 2020-12-13:
Those monitors seem to be notoriously hard to set-up. If you do a google search for "ASUS MB + Displaylink linux", you'll find a lot of issues that appear to remain unresolved. 

Maybe one of these issues contain useful information:
This config worked for a MB169B+ https://github.com/AdnanHodzic/displaylink-debian/issues/511#issuecomment-739472172
Or this one https://github.com/AdnanHodzic/displaylink-debian/issues/176

nd4c, 2020-12-13:
Please note, though, that the problem does not even involve the MB16AC yet...  I cannot install the displaylink driver on the laptop.  The laptop will not boot up. The MB16AC is not even plugged in yet.

However, yes, there seems to be issues with the MB16AC.  It was while searching for that solution that I found out about the DisplayLink drivers and ultimately about your scripts for debian.

nd4c, 2020-12-13:
I just tried to install proprietary AMD drivers for Ubuntu 20.04, but those failed to install.  This link was refering to the exact same hardware I have: https://askubuntu.com/questions/1272280/installing-proprietary-amd-drivers-on-ubuntu-20-04

nd4c, 2020-12-13:
I updated the Kernel to 5.8 and that has resolved the graphic driver issue, at least as far as now there is no warning about hardware exceleration.

nd4c, 2020-12-13:
And that was the problem...  OK! Solved!  I was able to reboot after installing displaylink-debian.  I will keep testing further and, of course, see if the MB16AC works...

nd4c, 2020-12-13:
Well, the MB16AC did not work.  But that's another issue...  This issue has been resolved.  Thanks for helping with this, and, of course, for your hard work maintaining these scripts!

nd4c, 2020-12-13:
OK, I followed the advice in the link you shared above: https://github.com/AdnanHodzic/displaylink-debian/issues/176#issuecomment-443434459.  My MB16AC is now working.  

I had a problem in that I was unable to "Move" the displays into a proper position - The MB was on the left and the Laptop was on the right.  And the Refresh Rates did not match: Laptop was 59.93 Hz (unchangeable) and the AUS 15" was 60 Hz.

I changed the AUS 15" refresh rate to match the Laptop (59.93 Hz) and now I can move the displays into position desired.

Also, note that the names are as follows:
Laptop(newline)eDP
AUS 15"(newline)DVI-I-1-1

Thank you!!!  Let me know if you need more details.

Barabazs, 2020-12-14:
Congratulations! I'm glad you were able to fix this. Can you post the output of  `sudo ./displaylink-debian.sh --debug` for future reference please?

I don't think the refresh rates have to match to be able to move the displays around. How did you try to move them?
I'm using ARandR to set everything up and then save the configuration as a script.

nd4c, 2020-12-14:
You are right about the refresh rates not needing to match.  Not sure what the problem was, but as soon as I got the display running, I opened the Linux Mint Display Settings gui.  At that point I was not able to "drag" the displays to a new location.  I would guess that there needed to be something "initialized", like a config file or something.  As soon as I edited the refresh rate and applied it, I was then able to move the screens.  I assumed (not good) that the problem was the refresh rates...

This is a copy of the original 20-displaylink.conf:
Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

This is the output of displaylink-debian.sh --debug:

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.8.0-33-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Renoir
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c2)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.6
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "DisplayLink"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x55 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0
Provider 1: id: 0xe6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xc4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xa2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x80 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

closed 2020-12-14 by Barabazs