AdnanHodzic/displaylink-debian #514: Support Elementary OS Hera

ummummhm, 2020-12-13:
This seems to install fine but does not take in hera

`DisplayLink driver for Debian GNU/Linux

[I]nstall
[U]ninstall

Select a key: [i/u]: i
Checking dependencies...
unzip is installed
linux-headers-5.4.0-58-generic is installed
dkms is installed
lsb-release is installed

------------------------------------------------------

Unsuported platform: elementary hera

This tool is Open Source and feel free to extend it
GitHub repo: https://goo.gl/6soXDE

------------------------------------------------------
`

Barabazs, 2020-12-13:
Please post the output of `sudo ./displaylink-debian.sh --debug`

ummummhm, 2020-12-13:
> Please post the output of `sudo ./displaylink-debian.sh --debug`

```
:~/displaylink-debian$ sudo ./displaylink-debian.sh --debug

DisplayLink driver for Debian GNU/Linux

[I]nstall
[U]ninstall

Select a key: [i/u]: I
Checking dependencies...
unzip is installed
linux-headers-5.4.0-58-generic is installed
dkms is installed
lsb-release is installed

------------------------------------------------------

Unsuported platform: elementary hera

This tool is Open Source and feel free to extend it
GitHub repo: https://goo.gl/6soXDE

------------------------------------------------------
```

The above is ALL i get when I try to run debug or I get  **command not found**

Barabazs, 2020-12-14:
You're using an old version of displaylink-debian. Please download the current version and provide the debug information.

closed 2020-12-14 by Barabazs