AdnanHodzic/displaylink-debian #517: MX Linux 19.3, Kernel 5.8.0-3-amd64 on Thinkpad L480 with Intel® UHD 620 Graphics

hashguru, 2020-12-18:
Tried all the suggestions so far, nothing worked. I have the Thinkpad USB 3.0 dock connected to my L480. Fresh install of MX linux with KDE. Legacy only boot with UEFI and secure boot disabled.

Two screens connected to the dock just remain black after hours of fiddling.
I am sure it is going to work, but my limited understanding of what to do next forces me to ask for help.

here follows the debug after fresh install of displaylink-debian to start the thread:

Barabazs, 2020-12-18:
You have to associate the providers. Try this procedure: https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection

hashguru, 2020-12-18:
i did try that, and many other things from closed issues. no go. i edited the original post to include the debug after associating providers.

Barabazs, 2020-12-18:
I would try to remove these files and see if it works after rebooting.
Don't forget to make a backup of them.

 /etc/X11/xorg.conf.d/synaptics.conf 
/etc/X11/xorg.conf.d/10-evdev.conf

hashguru, 2020-12-18:
Removed the files. Rebooted system. Associated providers. Display still black as night.

closed 2020-12-18 by Barabazs