AdnanHodzic/displaylink-debian #518: I can't connect second monitor with usb-c

pedromonici, 2020-12-21:
Hi! I'm testing displaylink but i can't get this works!
Here is my --debug:
`--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.9.0-4-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX150] (rev a1)
X11 version: 1.20.10-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 #Section "ServerLayout"
#    Identifier "layout"
#    Screen 0 "nvidia"
#    Inactive "intel"
#EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

#Section "Screen"
#    Identifier "intel"
#    Device "intel"
#EndSection

#Section "Device"
#    Identifier "nvidia"
#    Driver "nvidia"
#    Option "ConstrainCursor" "off"
#EndSection
#
#Section "Screen"
#    Identifier "nvidia"
#    Device "nvidia"
#    Option "AllowEmptyInitialConfiguration" "on"
#    Option "IgnoreDisplayDevices" "CRT"
#EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x46 cap: 0x2, Sink Output crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x17a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x158 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x136 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x114 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0xe2 cap: 0x5, Source Output, Source Offload crtcs: 0 outputs: 0 associated providers: 0 name:modesetting
`
I need to comment this lines to boot, because my system just broke in boot and then I can't boot anymore with this lines.

Barabazs, 2020-12-21:
What is the exact xrandr command that you tried?

pedromonici, 2020-12-21:
@Barabazs I tried every combination:
```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0
xrandr --setprovideroutputsource 5 0
```

Barabazs, 2020-12-21:
Try to replace your config with this:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
Reboot and retry the rest of the procedure please.

pedromonici, 2020-12-21:
@Barabazs Ok! Now i can do
``` xrandr --setprovideroutputsource ```
But my screen still don't shows up! Sorry foy my newbie experience!
Here is my xrandr --listproviders:
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 309mm x 174mm
   1920x1080     60.05 +  60.01*   59.97    59.96    59.93    48.04  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1440x900      59.89  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1360x768      59.80    59.96  
   1280x800      59.99    59.97    59.81    59.91  
   1152x864      60.00  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
DVI-I-5-4 disconnected (normal left inverted right x axis y axis)
DVI-I-4-3 disconnected (normal left inverted right x axis y axis)
DVI-I-3-2 disconnected (normal left inverted right x axis y axis)
DVI-I-2-1 disconnected (normal left inverted right x axis y axis)
```
And here is my --debug:
```
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.9.0-4-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX150] (rev a1)
X11 version: 1.20.10-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 #Section "ServerLayout"
#    Identifier "layout"
#    Screen 0 "nvidia"
#    Inactive "intel"
#EndSection

#Section "Device"
#    Identifier "intel"
#    Driver "modesetting"
#    Option "AccelMethod" "None"
#EndSection

Section "OutputClass"
        Identifier "DisplayLink"
        MatchDriver "evdi"
        Driver "modesetting"
        Option "AccelMode" "none"
EndSection

#Section "Device"
#    Identifier "nvidia"
#    Driver "nvidia"
#    Option "ConstrainCursor" "off"
#EndSection

#Section "Screen"
#    Identifier "nvidia"
#    Device "nvidia"
#    Option "AllowEmptyInitialConfiguration" "on"
#    Option "IgnoreDisplayDevices" "CRT"
#EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x18a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x168 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x146 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x124 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0xf2 cap: 0x5, Source Output, Source Offload crtcs: 0 outputs: 0 associated providers: 0 name:modesetting
```

Barabazs, 2020-12-21:
> ```
> xrandr --setprovideroutputsource 1 0
> xrandr --setprovideroutputsource 2 0
> xrandr --setprovideroutputsource 3 0
> xrandr --setprovideroutputsource 4 0
> ```

It looks like you forgot this: `xrandr --setprovideroutputsource 5 0` 
Try that first before we move on.

> Sorry for my newbie experience!

No need for apologies. We all have to start as beginners once.

pedromonici, 2020-12-21:
Only in that case:
```
xrandr --setprovideroutputsource 5 0
```
It me returns the same error:
```
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0xf2
  Serial number of failed request:  20
  Current serial number in output stream:  21
```

Barabazs, 2020-12-21:
Try  `xrandr --auto`

pedromonici, 2020-12-21:
@Barabazs Tried ```xrandr --auto``` and still not success

Barabazs, 2020-12-21:
Did you try this already?
```
Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection
```

pedromonici, 2020-12-21:
I tried now and my DVI ports still disconnects:
```
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 309mm x 174mm
   1920x1080     60.05*+  60.01    59.97    59.96    59.93    48.04  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1440x900      59.89  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1360x768      59.80    59.96  
   1280x800      59.99    59.97    59.81    59.91  
   1152x864      60.00  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
DVI-I-5-4 disconnected (normal left inverted right x axis y axis)
DVI-I-4-3 disconnected (normal left inverted right x axis y axis)
DVI-I-3-2 disconnected (normal left inverted right x axis y axis)
DVI-I-2-1 disconnected (normal left inverted right x axis y axis)
```

Here is my config file:
```
Section "Device"
    Identifier "DisplayLink"
    Driver "modesetting"
    Option "PageFlip" "false"
EndSection

Section "OutputClass"
	Identifier "DisplayLink"
	MatchDriver "evdi"
	Driver "modesetting"
	Option "AccelMode" "none"
EndSection

```

Barabazs, 2020-12-22:
Can you try with only this in your config?
```
Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection

```

pedromonici, 2020-12-22:
@Barabazs  Tried and still all DVI are disconnected

Barabazs, 2020-12-22:
I went through some old issues and I found another thing to try.
1. Restore the original configuration
2. Reboot with the docking station disconnected
3. Once it's fully booted you connect the docking
4. See if something happens with the monitor
5. If not, try the post install steps

pedromonici, 2020-12-22:
@Barabazs Ok, I restore the original file but then my computer don't boot (freezes at [OK] Started DisplayLink Driver Service). Then I remove this section in config:
```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection
```
The the computer boots, but after making the post install steps it didn't work yet

closed 2021-01-20 by Barabazs