AdnanHodzic/displaylink-debian #519: Install freezes laptop when performing post install steps

Dovydas5, 2020-12-22:
Installation setup freezes PC immediately when it starts performing post install steps. Need to force restart PC to use it again. 
Freezes last displaying : 
```Setup DisplayLink xorg.conf depending on graphics card```
After installation and force reboot debug log: 

```--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.8.0-7630-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile)
VGA (3D): NVIDIA Corporation GM108M [GeForce MX130] (rev a2)
X11 version: 1.20.9-2ubuntu1.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2c5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2a3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x281 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x25f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e1 cap: 0x2, Sink Output crtcs: 3 outputs: 6 associated providers: 1 name:modesetting

Barabazs, 2020-12-22:
Try to replace your /etc/X11/xorg.conf.d/20-displaylink.conf with this:

```
Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection
```

Dovydas5, 2020-12-22:
should i try to reinstall after a change?

Barabazs, 2020-12-22:
No, the install was successful. Just replace the config, reboot and retry the post-install steps.

Dovydas5, 2020-12-22:
[image] after restart stuck on this

Barabazs, 2020-12-22:
Is your laptop connected to the dock while booting?
If yes, disconnect and reboot. And only connect it once it's fully booted.

Dovydas5, 2020-12-22:
Tried with connected and disconected dock still the same

zettobelli85, 2021-01-19:
Hey! Tried your displaylink-installer too and my desktop freezed while performing post install steps.

After reboot linux mint, cinnamon got in any fallback-mode with graphical display errors. Then uninstalled your driver and after this.... it works! Don't know why,... do you have a guess? This behavoir seems odd to me.

Barabazs, 2021-01-20:
@zettobelli85 Please open a new issue.

closed 2021-01-20 by Barabazs