AdnanHodzic/displaylink-debian #523: instaslling displaylink on ubuntu 20.04

arpho, 2021-01-12:
Hi, I tried to install the driver following your guide, but the service does not start,this is what I get  with sudo ./displaylink-debian --debug:

`Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.8.0-36-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: radeon
Subsystem: /
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Oland [Radeon HD 8570 / R7 240/340 / Radeon 520 OEM]
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.1~20.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: File o directory non esistente
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x53 cap: 0x9, Source Output, Sink Offload crtcs: 2 outputs: 2 associated providers: 0 name:OLAND @ pci:0000:01:00.0
`
thank you in advance

Barabazs, 2021-01-12:
Can you try to run `sudo displaylink-debian.sh --install ` and copy the entire output here please.

arpho, 2021-01-12:
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.8.0-36-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at:
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]:

Downloading DisplayLink Ubuntu driver:

--2021-01-12 11:27:32--  https://www.displaylink.com/downloads/file?id=1576
Risoluzione di www.displaylink.com (www.displaylink.com)... 176.34.139.38,
52.214.244.113, 34.250.82.133, ...
Connessione a www.displaylink.com (www.displaylink.com)|176.34.139.38|:443...
connesso.
Richiesta HTTP inviata, in attesa di risposta... 303 See Other
Posizione:
https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1610447853&Signature=V%2Fr7FmE7mLkrWgRXarSf6lOif2M%3D
[segue]
--2021-01-12 11:27:33--
https://s3.eu-west-1.amazonaws.com/assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1610447853&Signature=V%2Fr7FmE7mLkrWgRXarSf6lOif2M%3D
Risoluzione di s3.eu-west-1.amazonaws.com (s3.eu-west-1.amazonaws.com)...
52.218.90.211
Connessione a s3.eu-west-1.amazonaws.com
(s3.eu-west-1.amazonaws.com)|52.218.90.211|:443...
connesso.
Richiesta HTTP inviata, in attesa di risposta... 200 OK
Lunghezza: 13889958 (13M) [application/octet-stream]
Salvataggio in: "DisplayLink_Ubuntu_5.3.1.zip"

DisplayLink_Ubuntu_5.3.1.zip

 100%[===============================================================================================================================================================================================>]
 13,25M  3,22MB/s    in 4,4s

2021-01-12 11:27:38 (3,00 MB/s) - "DisplayLink_Ubuntu_5.3.1.zip" salvato
[13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%

-------------------------------------------------------------------
Clone in 'evdi' in corso...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
[... 97 lines cut ...]
EVDI DKMS module already installed.
[[ Installing module configuration files ]]
[[ Installing EVDI library ]]
make: Nothing to be done for 'default'.
[ Installing x64-ubuntu-1604/DisplayLinkManager ]
[ Installing libraries ]
[ Installing firmware packages ]
[ Installing licence file ]
[ Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager sctripts ]

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu

Installation complete!

-------------------------------------------------------------------

Performing post install steps

Enable and start displaylink-driver service
Job for displaylink-driver.service failed because the control process
exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xe" for
details.

Barabazs, 2021-01-12:
>Job for displaylink-driver.service failed because the control process
exited with error code.
See "systemctl status displaylink-driver.service" and "journalctl -xe" for
details.

Did you look for additional info with:
`systemctl status displaylink-driver.service`
or
`journalctl -xe`

arpho, 2021-01-12:
yep

-- Logs begin at Sun 2021-01-10 12:44:27 CET, end at Tue 2021-01-12 16:56:22 CET. --
gen 12 16:48:56 Tacos sh[27092]: modprobe: ERROR: could not insert 'evdi': Operation not permitted
gen 12 16:48:56 Tacos systemd[1]: displaylink-driver.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Uscito processo unità
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Un processo ExecStartPre appartenente all'unità displaylink-driver.service è uscito.
-- 
-- Il codice di uscita del processo è 'exited' ed è uscito con 1.
gen 12 16:48:56 Tacos systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
-- Subject: Unit fallita
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unità displaylink-driver.service entrata nello stato 'failed' (fallito) con risultato 'exit-code'.
gen 12 16:48:56 Tacos systemd[1]: Failed to start DisplayLink Driver Service.
-- Subject: L'unità displaylink-driver.service è fallita
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unità displaylink-driver.service è fallita.
-- 
-- Il risultato è failed.
gen 12 16:48:56 Tacos kernel: Lockdown: modprobe: unsigned module loading is restricted; see man kernel_lockdown.7
gen 12 16:49:01 Tacos systemd[1]: displaylink-driver.service: Scheduled restart job, restart counter is at 133.
-- Subject: Il riavvio automatico di un'unità è stato schedulato
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Il riavvio automatico dell'unità displaylink-driver.service è stato schedulato, come risultato
-- delle impostazioni configurate in Restart=.
gen 12 16:49:01 Tacos systemd[1]: Stopped DisplayLink Driver Service.
-- Subject: L'unità displaylink-driver.service ha terminato la fase di spegnimento
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unità displaylink-driver.service ha terminato la fase di spegnimento.
gen 12 16:49:01 Tacos systemd[1]: Starting DisplayLink Driver Service...
-- Subject: L'unità displaylink-driver.service inizia la fase di avvio
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unità displaylink-driver.service ha iniziato la fase di avvio.
gen 12 16:49:01 Tacos sh[27232]: modprobe: ERROR: could not insert 'evdi': Operation not permitted
gen 12 16:49:01 Tacos kernel: Lockdown: modprobe: unsigned module loading is restricted; see man kernel_lockdown.7
gen 12 16:49:01 Tacos sh[27255]: Module evdi/1.7.2 already installed on kernel 5.8.0-36-generic/x86_64
gen 12 16:49:01 Tacos kernel: Lockdown: modprobe: unsigned module loading is restricted; see man kernel_lockdown.7
gen 12 16:49:01 Tacos sh[27233]: modprobe: ERROR: could not insert 'evdi': Operation not permitted
gen 12 16:49:01 Tacos systemd[1]: displaylink-driver.service: Control process exited, code=exited, status=1/FAILURE
-- Subject: Uscito processo unità
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Un processo ExecStartPre appartenente all'unità displaylink-driver.service è uscito.
-- 
-- Il codice di uscita del processo è 'exited' ed è uscito con 1.
gen 12 16:49:01 Tacos systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.
-- Subject: Unit fallita
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unità displaylink-driver.service entrata nello stato 'failed' (fallito) con risultato 'exit-code'.
gen 12 16:49:01 Tacos systemd[1]: Failed to start DisplayLink Driver Service.
-- Subject: L'unità displaylink-driver.service è fallita
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
[... 1115 lines cut ...]
-- 
-- L'unità displaylink-driver.service è fallita.
-- 
-- Il risultato è failed.
gen 12 16:51:07 Tacos kernel: usb 4-5: reset full-speed USB device number 2 using ohci-pci
gen 12 16:51:12 Tacos systemd[1]: displaylink-driver.service: Scheduled restart job, restart counter is at 158.
-- Subject: Il riavvio automatico di un'unità è stato schedulato
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Il riavvio automatico dell'unità displaylink-driver.service è stato schedulato, come risultato
-- delle impostazioni configurate in Restart=.
gen 12 16:51:12 Tacos systemd[1]: Stopped DisplayLink Driver Service.
-- Subject: L'unità displaylink-driver.service ha terminato la fase di spegnimento
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unità displaylink-driver.service ha terminato la fase di spegnimento.
gen 12 16:51:12 Tacos systemd[1]: Starting DisplayLink Driver Service...
-- Subject: L'unità displaylink-driver.service inizia la fase di avvio
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- L'unità displaylink-driver.service ha iniziato la fase di avvio.
gen 12 16:51:12 Tacos sh[31061]: modprobe: ERROR: could not insert 'evdi'

Barabazs, 2021-01-12:
And?

arpho, 2021-01-12:
sory I have attached two file but one of them was too big; output of
systemctl:
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/lib/systemd/system/displaylink-driver.service;
enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue
2021-01-12 16:55:30 CET; 4s ago
    Process: 38318 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install
$(ls -t /usr/src | grep evdi | head -n1  | sed -e "s:-:/:") && modprobe
evdi) (code=exited, status=1/FAILURE)

I do not know how to send the journalctl -xe is here:
https://pastebin.com/Z0hSQtLv

Barabazs, 2021-01-14:
> sory I have attached two file but one of them was too big; 

Attachments via mail do not work.
Is secureboot disabled?

arpho, 2021-01-14:
thank you I disabled secureboot, uninstalled and installed again, now
xrandr --listproviders returns 5 providers, so I think it is properly
installed now

closed 2021-01-14 by Barabazs