AdnanHodzic/displaylink-debian #524: Support Linux Mint 20.1

vama22, 2021-01-13:
--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyssa
Kernel: 5.4.0-60-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
VGA (3D): NVIDIA Corporation GP107GLM [Quadro P620] (rev a1)
X11 version: 1.20.9-2ubuntu1.1~20.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x218 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x241 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 6 associated providers: 1 name:modesetting

Barabazs, 2021-01-15:
@vama22 Please let me know if it works for you.

closed 2021-01-15 by Barabazs

vama22, 2021-01-18:
Hi, it works very well :) Thanks