AdnanHodzic/displaylink-debian #525: Support LMDE 4 with this configuration

vama22, 2021-01-13:
--------------- Linux system info ----------------

Distro: Linuxmint
Release: debbie
Kernel: 4.19.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
VGA (3D): NVIDIA Corporation Device 1cbd (rev a1)
X11 version: 1.20.4-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x68 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 6 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Barabazs, 2021-01-15:
It's already supported.
What issue are you experiencing?

vama22, 2021-01-18:
Hi, it's related to this case #508, after installation it's impossible to reboot. Since that time, I tried to upgrade bios and all firmwares and also to upgrade to newer kernel version, but nothing resolves the issue. On the same machine, Ubuntu works fine, Linux Mint 20.1 also, but only LDME 4 doens't work.

Barabazs, 2021-01-18:
Without logs, there is not much I can do.

vama22, 2021-01-19:
I'm able to give you all logs you want, which one would you like? A big part of them are in the case #508.

Barabazs, 2021-01-20:
The output when installing and the boot log.

vama22, 2021-01-22:
Hi, in attachment the logs you asked. Don't hesitate to indicate me which specific log files you would.
[Install.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5855308/Install.txt)
[boot.log](https://github.com/AdnanHodzic/displaylink-debian/files/5855314/boot.log)
[kern.log](https://github.com/AdnanHodzic/displaylink-debian/files/5855315/kern.log)
[messages.log](https://github.com/AdnanHodzic/displaylink-debian/files/5855320/messages.log)
[syslog.log](https://github.com/AdnanHodzic/displaylink-debian/files/5855321/syslog.log)

Barabazs, 2021-01-22:
Thanks @vama22 .
As far as I can see/understand, your laptop boots successfully?

vama22, 2021-01-22:
Unfortunately no, the laptop blocks with the message: iwlwifi BIOS contains WGDS but no WRDS. The laptop model is Dell Precision 3541.

Barabazs, 2021-01-23:
But after that message it continues to boot and doesn't appear to block anything?

vama22, 2021-01-25:
The screen remanis blocked on this message also if I wait a long. The login screen never appears.

Barabazs, 2021-01-25:
Are you booting with USB devices connected? (If yes, try to boot with nothing connected)
According to the install log, you have a lot of upgradable packages. Maybe try to upgrade those first.

vama22, 2021-01-25:
I tried, but I've the same result. In attachment the new install log file.
[Install.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5865917/Install.txt)

vama22, 2021-01-25:
Also, on the error screen, with CRTL+ALT+F1 I can open a shell and login with my user session, but without GUI. Perhaps something related to "plymouth" ?!

Barabazs, 2021-01-25:
Can you post the output of `sudo ./displaylink-debian.sh --debug`?

vama22, 2021-01-25:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: debbie
Kernel: 4.19.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
VGA (3D): NVIDIA Corporation Device 1cbd (rev a1)
X11 version: 1.20.4-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x68 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 6 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2021-01-25:
Did you uninstall it?

vama22, 2021-01-25:
Sorry, the output from the shell without GUI is "Can't open display".

I followed the new case #529 by removing 20-displaylink.conf and I'm able to boot again. In attachment the 20-displaylink.conf and Debug files.
[20-displaylink.conf.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5868473/20-displaylink.conf.txt)
[Debug.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5868474/Debug.txt)

Barabazs, 2021-01-25:
Can you try with this config?
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

vama22, 2021-01-26:
The error message appears, but it doesn't block the screen and so I can login. However the screens are not recognized.

If I boot with USB devices connected I have an error, see attachement, about plymouth and after some minutes I can login.
[plymouth-start.service.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5874890/plymouth-start.service.txt)

Barabazs, 2021-01-26:
> The error message appears, but it doesn't block the screen and so I can login. However the screens are not recognized.

Can you give the output of `sudo ./displaylink-debian.sh --debug`?

vama22, 2021-01-27:
Sure
[Debug.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5878724/Debug.txt)

Barabazs, 2021-01-27:
You need to associate the providers:
[post install guide/display-detection](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection)

vama22, 2021-01-27:
It works, it's wondeful :) Many many thanks.

Barabazs, 2021-01-27:
> It works, it's wondeful :) Many many thanks.

Glad it worked out! :+1: 

>How just I can set display-detection forever, without having to do the same after each reboot? Is only possible with alias?

Alias, or you could try to write a script to automate it.

closed 2021-01-27 by Barabazs