AdnanHodzic/displaylink-debian #528: Debian Buster - Targus Dock - Screens detected but remains black or frozen (with wrong display)

rrajkomar, 2021-01-16:
Hi,
Recently switched to more recent machine and had to buy a displaylink dock to go with it.
I chose a Targus with 4 connected screens.
While the display link install seems to have gone smootly, the setup is not entirely functional as the four screens upon their respective provider activation remain black or sometimes displays a screen with nothing that can be moved.

Here's the debug output :
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: d

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.8.0-0.bpo.2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation Device 9bc4 (rev 05)
NVIDIA Corporation Device 1e93 (rev a1)
VGA (3D): 
X11 version: 1.20.4-1+deb10u2
X11 configs: /etc/X11/xorg.conf.d/30-displaylink.conf /etc/X11/xorg.conf.d/20-intel.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcd cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 1 name:Intel
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Here are the related file contents

20-intel.conf
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
    BusID		"0:2:0"
EndSection
```

30-displaylink.conf
```
Section "Device"
    Identifier  "DisplayLink"
    Driver "modesetting"
    Option "PageFlip" "false"
EndSection
```
Any help would be greatly appreciated as I don't get why it's not working

Barabazs, 2021-01-16:
Have you tried changing the priority of your displaylink.conf?

rrajkomar, 2021-01-16:
Yes I tried changing the order by putting the displaylink file with the
"10-" prefix. While that didn't change the boot, it prevented me from
running the xrandr command to setprovideroutputsource because provider 0
and provider 1 get modesetting

Barabazs, 2021-01-17:
> -------------------- Monitors --------------------
>
> Providers: number : 5
> Provider 0: id: 0xcd cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 1 [...]

It looks like you did not associate the providers like described here:
https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection

rrajkomar, 2021-01-17:
In that specific output I had only ran the first setproviderputsource
command in order to test out configurations with one monitor to find one
that works.
But the issue is the same if I run all four commands.
I'll post another output with all four later (no access to the machine ATM)

rrajkomar, 2021-01-17:
P.S. : The dock model is the Targus DOCK570EUZ

Barabazs, 2021-01-17:
In that case you can ignore my question.

Another thing to try:
try disabling/removing your 20-intel.conf and see what that gives when rebooting, setting the providers, etc.

rrajkomar, 2021-01-17:
If I disable the 20-intel.conf file here's what happens : 
```
xrandr --listproviders
Providers: number : 5
Provider 0: id: 0xcb cap: 0x2, Sink Output crtcs: 3 outputs: 1 associated providers: 0 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

And from that point if I try to set provideroutputsource I get : 
```
xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0xcb
  Serial number of failed request:  19
  Current serial number in output stream:  20
```

rrajkomar, 2021-01-18:
Ok So, this morning I had an idea : why not try to start the system using the nvidia card as base provider instead of the intel one.
And what do you know it helps somehow.
When I boot on the nvidia first with the following configs : 
10-nvidia.conf
```
Section "Device"
	Identifier "Nvidia"
	Driver "nvidia"
	BusID "1:0:0"
EndSection
```

20-intel.conf
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
    BusID		"0:2:0"
EndSection
```

30-displaylink.conf
```
Section "Device"
    Identifier  "DisplayLink"
    Driver "modesetting"
    Option "PageFlip" "false"
EndSection
```

I am able to start the external displays but there are a few kinks :
- I can't use the main monitor (laptop screen) it is detected but doesn't see to display anything
- I am forced to use an additionnal HDMI cable to connect of the external screen to the HDMI port of the laptop (which I believe is connected to the NVIDIA card)

I'm also confused about one thing : does the number of modesetting (displaylink) providers need to matche the number of displaylink screen or the number of displaylink devices ?
(Had a weird issue earlier that caused to screen to turn on when I only activated one with xrandr)

Now the ideal solution would be for me to be able to boot on the intel card with the nvidia active as if there was a monitor connected  so that I can then use the NVIDIA card as the renderer for all screens

Here the ourput of providers as of right now
(only the laptop screen black with mouse visible and the DVI-5-4 screen turn off as it the same monitor as the HDMI-0 which is connected to my HDMI port) : 
```
xrandr --listproviders 
Providers: number : 6
Provider 0: id: 0x269 cap: 0x1, Source Output crtcs: 4 outputs: 6 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0xd0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xae cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x45 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 1 name:Intel
Provider 4: id: 0x6a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x8c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

Any thoughts are welcomed.

Barabazs, 2021-01-18:
Some random observations/questions:
* Are you using X11 or Wayland?
* Have you tried to let xorg autoconfigure?
* IIRC the closed source NVIDIA driver does not work with displaylink driver.

rrajkomar, 2021-01-18:
AFAIK I'm using X11

What to you mean by let it autoconfigure ? you mean no conf files and just go like that. No I haven't, I can try.

As far as NVIDIA / DisplayLink I think so too. Because the dock I previously had worked with the NVIDIA directly (needed the NVIDIA card with that dock anyway for the fourth screen) but now that I have a four screen dock I'd rather not plug a cable on the HDMi port just for it to kinda work.

rrajkomar, 2021-01-19:
Ok so this morning, I was able to try other setups : 

- with no xorg conf file > does not work : I boot on the intel card with modesetting driver and therefore can't run any setoutputprovider commands

- without the displaylink part (uninstalled it reboot it and try with only the NVIDIA system [nvidia prime]) : did not work the monitors detected by the nvidia card are not linked to the targus dock

Now I'm back to my previous setup : 
Hdmi cable plugged on the nvidia (main monitor used to boot)
When running the xrandr commands, all monitors start but the laptop monitor is black (only mouse visible)

FYI:
```
xrandr -q | grep connected
DP-0 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
HDMI-0 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 521mm x 293mm
DP-4 disconnected (normal left inverted right x axis y axis)
DVI-I-5-4 connected (normal left inverted right x axis y axis)
DVI-I-4-3 connected 1920x1080+1920+0 (normal left inverted right x axis y axis) 521mm x 293mm
DVI-I-3-2 connected 1920x1080+7680+0 (normal left inverted right x axis y axis) 521mm x 293mm
DVI-I-2-1 connected 1920x1080+5760+0 (normal left inverted right x axis y axis) 521mm x 293mm
eDP1 connected 1920x1080+3840+0 (normal left inverted right x axis y axis) 340mm x 190mm
VIRTUAL1 disconnected (normal left inverted right x axis y axis)
```
HDMI-0 and the DP-* are the nvidia monitors
DVI-* are the displaylink monitors
eDP1 is the laptop monitor
VIRTUAL1 is something else (never used, don't know how to anyway)

Barabazs, 2021-01-23:
Maybe you could try to get your setup working piece by piece
For example: 
- start with only one monitor connected to the dock
- with only the intel and displaylink config active

And then try to build up, device per device.

rrajkomar, 2021-02-05:
Sorry it took me a while to get back to this. 
Tried booting only a single monitor with the intel and displaylink config only.
The same issue happens.

closed 2021-03-06 by Barabazs