AdnanHodzic/displaylink-debian #529: MX-19.3 -  permanent black screen

EmanueleCarrieri87, 2021-01-25:
Hi,

i installed DL on my notebook (Dell Inspiron 5593) and MX-Linux 19.3 (KDE), the installation looks good but after reboot 
and after loading the OS i have a permanent black screen.

The only way I've found to be able to log in again is to boot with a live usb and manually delete the config file: /etc/X11/xorg.conf.d/20-displaylink.conf, so after the reboot the system is available again.

I have already tried leaving in the configuration directory only the file: 20-displaylink.conf

Below is the debug log (obviously the 20-displaylink.conf file is not found, since I had to delete it to access the system again)
```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.6.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 8a52 (rev 07)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX230] (rev ff)
X11 version: 1.20.9-2~mx19ahs+1
X11 configs: /etc/X11/xorg.conf.d/synaptics.conf /etc/X11/xorg.conf.d/10-evdev.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: File not found
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x44 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x149 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x127 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x105 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xe3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------
```

I am available for other tests / info
Thanks

Barabazs, 2021-01-25:
Can you post the content of these files? 
/etc/X11/xorg.conf.d/synaptics.conf /etc/X11/xorg.conf.d/10-evdev.conf

EmanueleCarrieri87, 2021-01-25:
Sure

**synaptics.conf** 
```
Section "InputClass"
        Identifier      "touchpad catchall"             # required
        MatchIsTouchpad "on"                            # required
        Driver          "synaptics"                     # required
        Option          "MinSpeed"              "0.5"
        Option          "MaxSpeed"              "1.0"
        Option          "AccelFactor"           "0.075"
        Option          "TapButton1"            "1"
        Option          "TapButton2"            "2"     # multitouch
        Option          "TapButton3"            "3"     # multitouch
        Option          "VertTwoFingerScroll"   "1"     # multitouch
        Option          "HorizTwoFingerScroll"  "1"     # multitouch
        Option          "VertEdgeScroll"        "1"
        Option          "HorizEdgeScroll"       "1" 
        Option          "CoastingSpeed"         "8"
        Option          "CornerCoasting"        "1"
        Option          "CircularScrolling"     "1"
        Option          "CircScrollTrigger"     "7"
        Option          "EdgeMotionUseAlways"   "1"
        Option          "LBCornerButton"        "8"     # browser "back" btn
        Option          "RBCornerButton"        "9"     # browser "forward" btn
EndSection
```

**10-evdev.conf** 
```
#
# Catch-all evdev loader for udev-based systems
# We don't simply match on any device since that also adds accelerometers
# and other devices that we don't really want to use. The list below
# matches everything but joysticks.

Section "InputClass"
        Identifier "evdev pointer catchall"
        MatchIsPointer "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection

Section "InputClass"
        Identifier "evdev keyboard catchall"
        MatchIsKeyboard "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection

Section "InputClass"
        Identifier "evdev touchpad catchall"
        MatchIsTouchpad "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection

#Section "InputClass"
#        Identifier "evdev tablet catchall"
#        MatchIsTablet "on"
#        MatchDevicePath "/dev/input/event*"
#        Driver "evdev"
#EndSection

Section "InputClass"
        Identifier "evdev touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection

```

Barabazs, 2021-01-25:
Do you know what was in the displaylink config?

Please try with this config:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

EmanueleCarrieri87, 2021-01-25:
it works !
thank you very much

for troubleshooting, this is the post install config file:

```
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

```

thank you again, bye

closed 2021-01-25 by Barabazs