AdnanHodzic/displaylink-debian #531: Only blinks the power button. Logs are full of "unhandled action 'bind' on /sys/devices..."

alanjds, 2021-01-26:
Product: ASUSTEK MB169B+

`/etc/X11/xorg.conf.d/20-displaylink.conf`:
```
$ cat /etc/X11/xorg.conf.d/20-displaylink.conf 
#Section "OutputClass"
#    Identifier  "DisplayLink"
#    MatchDriver "evdi"
#    Driver      "modesetting"
#    Option      "AccelMethod" "none"
#EndSection

Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

#Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection
```
When plugged, the power button on the device cycles between orange (1sec), green (<300ms), black (~1.5sec)

The screen is kept black at all times.

And `journalctl -f` yields in loop:
```
jan 26 20:33:25 lenovopill kernel: usb 2-4: new SuperSpeed Gen 1 USB device number 107 using xhci_hcd
jan 26 20:33:25 lenovopill kernel: usb 2-4: New USB device found, idVendor=17e9, idProduct=ff0b, bcdDevice=31.13
jan 26 20:33:25 lenovopill kernel: usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jan 26 20:33:25 lenovopill kernel: usb 2-4: Product: ASUSTEK MB169B+
jan 26 20:33:25 lenovopill kernel: usb 2-4: Manufacturer: DisplayLink
jan 26 20:33:25 lenovopill kernel: usb 2-4: SerialNumber: J9LMTF014868
jan 26 20:33:25 lenovopill mtp-probe[20211]: checking bus 2, device 107: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4"
jan 26 20:33:25 lenovopill mtp-probe[20211]: bus: 2, device: 107 was not an MTP device
jan 26 20:33:26 lenovopill systemd[1]: Starting DisplayLink Driver Service...
jan 26 20:33:26 lenovopill systemd[1]: Started DisplayLink Driver Service.
jan 26 20:33:26 lenovopill upowerd[1867]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 26 20:33:26 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0
jan 26 20:33:26 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.1
jan 26 20:33:26 lenovopill kernel: usb 2-4: USB disconnect, device number 107
jan 26 20:33:26 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 26 20:33:26 lenovopill systemd[1]: Stopping DisplayLink Driver Service...
```

USB cable is good. Tested with other USB3 devices on the same port.

Barabazs, 2021-01-27:
Please post the output of `sudo ./displaylink-debian.sh --debug`

alanjds, 2021-01-27:
```
--------------- Linux system info ----------------

Distro: Ubuntu                                                                                                                 
Release: bionic                                                                                                                
Kernel: 5.4.0-62-generic                                                                                                       
                                                                                                                               
---------------- DisplayLink info ----------------                                                                             
                                                                                                                               
Driver version: 1.7.2                                                                                                          
4.4.24                                                                                                                         
5.1.26
5.2.14
DisplayLink service status: 
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2~18.04.4
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 #Section "OutputClass"
#    Identifier  "DisplayLink"
#    MatchDriver "evdi"
#    Driver      "modesetting"
#    Option      "AccelMethod" "none"
#EndSection

Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

#Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x152 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x130 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x10e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xec cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Barabazs, 2021-01-27:
Can you try to replace your displaylink config with this:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

```

alanjds, 2021-01-27:
Sure.

Already tried with this one before. Lets try again.

alanjds, 2021-01-27:
Same result: Blinking orange on power button, no signal on screen.

dmesg -w
```

[73084.371988] usb 1-4: USB disconnect, device number 81
[73110.271738] usb 1-4: new high-speed USB device number 82 using xhci_hcd
[73111.887706] usb 1-4: new high-speed USB device number 83 using xhci_hcd
[73112.036821] usb 1-4: New USB device found, idVendor=17e9, idProduct=ff0b, bcdDevice=31.13
[73112.036829] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[73112.036834] usb 1-4: Product: ASUSTEK MB169B+
[73112.036838] usb 1-4: Manufacturer: DisplayLink
[73112.036842] usb 1-4: SerialNumber: J9LMTF014868
[73112.048944] usb 1-4: USB disconnect, device number 83
[73113.647777] usb 1-4: new high-speed USB device number 84 using xhci_hcd
[73115.259730] usb 1-4: new high-speed USB device number 85 using xhci_hcd
[73116.879554] usb 1-4: new high-speed USB device number 86 using xhci_hcd
[73118.495521] usb 1-4: new high-speed USB device number 87 using xhci_hcd
[73120.111715] usb 1-4: new high-speed USB device number 88 using xhci_hcd
[73121.735518] usb 1-4: new high-speed USB device number 89 using xhci_hcd
[73123.359640] usb 1-4: new high-speed USB device number 90 using xhci_hcd
[73124.979352] usb 1-4: new high-speed USB device number 91 using xhci_hcd
[73126.591442] usb 1-4: new high-speed USB device number 92 using xhci_hcd
[73126.740621] usb 1-4: New USB device found, idVendor=17e9, idProduct=ff0b, bcdDevice=31.13
[73126.740630] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[73126.740634] usb 1-4: Product: ASUSTEK MB169B+
[73126.740639] usb 1-4: Manufacturer: DisplayLink
[73126.740642] usb 1-4: SerialNumber: J9LMTF014868
[73126.752712] usb 1-4: USB disconnect, device number 92
...
[73137.119310] usb 1-4: new high-speed USB device number 98 using xhci_hcd
[73137.279591] usb 1-4: device descriptor read/all, error -71
[73138.875451] usb 1-4: new high-speed USB device number 100 using xhci_hcd
...
[73173.167583] usb 1-4: USB disconnect, device number 120
[73174.758795] usb 1-4: new high-speed USB device number 121 using xhci_hcd
[73174.886837] usb 1-4: Device not responding to setup address.
[73175.094800] usb 1-4: Device not responding to setup address.
[73175.302785] usb 1-4: device not accepting address 121, error -71
[73178.034760] usb 1-4: new high-speed USB device number 123 using xhci_hcd
```

And keep going on this.

alanjds, 2021-01-27:
This is new:

```
an 27 16:21:37 lenovopill mtp-probe[17252]: checking bus 2, device 52: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4"
jan 27 16:21:37 lenovopill mtp-probe[17252]: bus: 2, device: 52 was not an MTP device
jan 27 16:21:37 lenovopill systemd[1]: displaylink-driver.service: Start request repeated too quickly.
jan 27 16:21:37 lenovopill systemd[1]: displaylink-driver.service: Failed with result 'start-limit-hit'.
jan 27 16:21:37 lenovopill systemd[1]: Failed to start DisplayLink Driver Service.
jan 27 16:21:37 lenovopill upowerd[1867]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 27 16:21:37 lenovopill kernel: usb 2-4: USB disconnect, device number 52
```

complete `journalctl -f`:
```
jan 27 16:21:27 lenovopill kernel: usb 2-4: new SuperSpeed Gen 1 USB device number 46 using xhci_hcd
jan 27 16:21:27 lenovopill kernel: usb 2-4: New USB device found, idVendor=17e9, idProduct=ff0b, bcdDevice=31.13
jan 27 16:21:27 lenovopill kernel: usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jan 27 16:21:27 lenovopill kernel: usb 2-4: Product: ASUSTEK MB169B+
jan 27 16:21:27 lenovopill kernel: usb 2-4: Manufacturer: DisplayLink
jan 27 16:21:27 lenovopill kernel: usb 2-4: SerialNumber: J9LMTF014868
jan 27 16:21:27 lenovopill mtp-probe[16717]: checking bus 2, device 46: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4"
jan 27 16:21:27 lenovopill mtp-probe[16717]: bus: 2, device: 46 was not an MTP device
jan 27 16:21:27 lenovopill systemd[1]: Starting DisplayLink Driver Service...
jan 27 16:21:27 lenovopill systemd[1]: Started DisplayLink Driver Service.
jan 27 16:21:27 lenovopill upowerd[1867]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4

jan 27 16:21:28 lenovopill kernel: usb 2-4: USB disconnect, device number 46
jan 27 16:21:28 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0
jan 27 16:21:28 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.1
jan 27 16:21:28 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 27 16:21:28 lenovopill systemd[1]: Stopping DisplayLink Driver Service...
jan 27 16:21:29 lenovopill systemd[1]: Stopped DisplayLink Driver Service.
jan 27 16:21:29 lenovopill kernel: usb 2-4: new SuperSpeed Gen 1 USB device number 47 using xhci_hcd
jan 27 16:21:29 lenovopill kernel: usb 2-4: New USB device found, idVendor=17e9, idProduct=ff0b, bcdDevice=31.13
jan 27 16:21:29 lenovopill kernel: usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jan 27 16:21:29 lenovopill kernel: usb 2-4: Product: ASUSTEK MB169B+
jan 27 16:21:29 lenovopill kernel: usb 2-4: Manufacturer: DisplayLink
jan 27 16:21:29 lenovopill kernel: usb 2-4: SerialNumber: J9LMTF014868
jan 27 16:21:29 lenovopill mtp-probe[16806]: checking bus 2, device 47: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4"
jan 27 16:21:29 lenovopill mtp-probe[16806]: bus: 2, device: 47 was not an MTP device
jan 27 16:21:29 lenovopill systemd[1]: Starting DisplayLink Driver Service...
jan 27 16:21:29 lenovopill systemd[1]: Started DisplayLink Driver Service.
jan 27 16:21:29 lenovopill upowerd[1867]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 27 16:21:29 lenovopill kernel: usb 2-4: usbfs: usb_submit_urb returned -19
jan 27 16:21:29 lenovopill kernel: usb 2-4: usbfs: usb_submit_urb returned -19
jan 27 16:21:30 lenovopill kernel: usb 2-4: USB disconnect, device number 47
jan 27 16:21:30 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 27 16:21:30 lenovopill systemd[1]: Stopping DisplayLink Driver Service...
jan 27 16:21:30 lenovopill systemd[1]: Stopped DisplayLink Driver Service.
jan 27 16:21:30 lenovopill kernel: usb 2-4: new SuperSpeed Gen 1 USB device number 48 using xhci_hcd
jan 27 16:21:30 lenovopill kernel: usb 2-4: New USB device found, idVendor=17e9, idProduct=ff0b, bcdDevice=31.13
jan 27 16:21:30 lenovopill kernel: usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
jan 27 16:21:30 lenovopill kernel: usb 2-4: Product: ASUSTEK MB169B+
jan 27 16:21:30 lenovopill kernel: usb 2-4: Manufacturer: DisplayLink
jan 27 16:21:30 lenovopill kernel: usb 2-4: SerialNumber: J9LMTF014868
jan 27 16:21:30 lenovopill mtp-probe[16884]: checking bus 2, device 48: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4"
jan 27 16:21:30 lenovopill mtp-probe[16884]: bus: 2, device: 48 was not an MTP device
jan 27 16:21:30 lenovopill systemd[1]: Starting DisplayLink Driver Service...
jan 27 16:21:30 lenovopill systemd[1]: Started DisplayLink Driver Service.
jan 27 16:21:30 lenovopill upowerd[1867]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4

jan 27 16:21:31 lenovopill kernel: usb 2-4: USB disconnect, device number 48
jan 27 16:21:31 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0
jan 27 16:21:31 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.1
jan 27 16:21:31 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 27 16:21:31 lenovopill systemd[1]: Stopping DisplayLink Driver Service...
jan 27 16:21:32 lenovopill systemd[1]: Stopped DisplayLink Driver Service.
[... cut ...]
jan 27 16:21:35 lenovopill systemd[1]: displaylink-driver.service: Failed with result 'start-limit-hit'.
jan 27 16:21:35 lenovopill systemd[1]: Failed to start DisplayLink Driver Service.
jan 27 16:21:35 lenovopill upowerd[1867]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
jan 27 16:21:35 lenovopill kernel: usb 2-4: USB disconnect, device number 51
jan 27 16:21:35 lenovopill upowerd[1867]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4
```

Barabazs, 2021-01-28:
You're not the first to have a hard time getting this type of monitor to work with displaylink.

But as far as displaylink-debian is concerned, everything looks fine and the install was successful.
You'll have to try different configurations to see if it works or not. Also try to see if the monitor works on a Windows machine.

closed 2021-01-28 by Barabazs

alanjds, 2021-01-29:
Thanks.

I do not have a Windows machine with a USB3 port. If I could get one, which version of Windows would you suggest?

Barabazs, 2021-01-31:
It's just for testing the monitor, as most DisplayLink devices work out of the box on Windows.
There might be other/better ways to test..

alanjds, 2021-01-31:
I see.

The monitor is fine. I bought from a friend that was using it on his Ubuntu
(I saw it working). Told me that it is good, yet a PITA to setup 🙈.

Anyway will try some Windows if a notebook with USB3 stops by.