AdnanHodzic/displaylink-debian #533: Issue on a newer machine running ubuntu 20.04

Hypnocrit, 2021-02-01:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.8.0-41-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation Device 249d (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 6 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x30f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2ed cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2cb cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e6 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

I did follow from what I can tell all the instructions.  I did also redirect the providers to zero as you can see above.  I am not sure what i can do about this problem.  No serious rush.. but I will donate when this is resolved as a promise. 

Machine is a GP76 Leopard.  
Hardware runs entirely same on Windows just fine.   
The video card is a 3070.  
Other monitors work with the computer just fine, the 2 onboard slots do work.
At the time of review, all other external monitors are disabled, and I have just startech device connected and also external mouse.
I will gladly give any additional details you like.  I will do my best to respond in timely fashion.

Barabazs, 2021-02-01:
Could you try:
```
xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```
And then post the output of `xrandr --listproviders`

Hypnocrit, 2021-02-02:
lance@lance-laptop:~$ xrandr --setprovideroutputsource modesetting NVIDIA-0
lance@lance-laptop:~$ xrandr --auto
lance@lance-laptop:~$ xrandr --listproviders
Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 6 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x324 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x302 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2e0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2be cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x20a cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 1 associated providers: 1 name:modesetting
lance@lance-laptop:~$ 

Thanks for your time. 

I had some confusion around the nvidia drivers potentially, im reading elsewhere that the nvidia proprietary drivers may not work.  that seems odd seem as how that is the only drivers i can install.  Running nvidia-460 i believe it is.  I want to make sure this isnt a base package problem.  I was unable to get the normal default drivers to work also and then found this and figured I would try it.  The windows side of things works great, no questions asked.

Barabazs, 2021-02-02:
You'll have to use the open source drivers.
[Source](https://support.displaylink.com/knowledgebase/articles/641668-known-issues-with-displaylink-ubuntu-support)

Since you're on Ubuntu, you could try to install the official DisplayLink package. (but also with the open source drivers for your GPU)
First uninstall with `sudo ./displaylink-debian.sh --uninstall`, then reboot, Install the latest from https://www.displaylink.com/downloads/ubuntu

BenMueller, 2021-02-15:
Issue seems to be upstream, 20.04 is broken. https://displaylink.org/forum/showthread.php?t=67148

closed 2021-02-23 by Barabazs