AdnanHodzic/displaylink-debian #534: Unable to boot on Ubuntu 20.04 after install

bastoune, 2021-02-05:
After a fresh re-install of Ubuntu 20.04, I am able to install DisplayLink but if I reboot, I am stuck on a black screen with a blinking underscore.
I can login with `CTRL+Alt+F2` and then the only way to boot again is to remove DisplayLink with `sudo displaylink-installer uninstall`

It seems I have troubles with Xorg.
I can see multiple crash reports about xorg but don't know how to read those:

```
bastio@bastio-Latitude-7400:/var/crash$ ll
total 8700
drwxrwsrwt  3 root     whoopsie    4096 févr.  5 09:53 ./
drwxr-xr-x 14 root     root        4096 juil. 31  2020 ../
drwxrwsr-x  3 bastio   whoopsie    4096 févr.  5 09:48 system/
-rw-r-----  1 bastio   whoopsie 8891980 févr.  5 09:55 _usr_lib_xorg_Xorg.1000.crash
-rw-rw-r--  1 bastio   whoopsie       0 févr.  5 09:42 _usr_lib_xorg_Xorg.1000.upload
-rw-------  1 whoopsie whoopsie      37 févr.  5 09:42 _usr_lib_xorg_Xorg.1000.uploaded

```

Could it be related to an installation of Barrier ?

Labtop: Dell Latitude 7400

Here is the debug after a reboot and so after an uninstallation :
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.8.0-41-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Here after a re-installation

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.8.0-41-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x147 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x125 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x101 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

Barabazs, 2021-02-06:
Please post some logs (boot, dmesg,...).

bastoune, 2021-02-11:
I close the topic, I succeed to install it by removing "barrier"

closed 2021-02-11 by bastoune

aelray, 2021-07-05:
what was the "barrier"?

bastoune, 2021-07-07:
https://github.com/debauchee/barrier