AdnanHodzic/displaylink-debian #535: Using displaylink is not possible, very slow, not possible to adjust monitor

goyon2000, 2021-02-07:
Using displaylink is not possible, very slow, not possible to adjust monitor

i do
[debug.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5939424/debug.txt)
[code]
System:    Host: yvc-3000-N200 Kernel: 5.4.0-65-generic x86_64 bits: 64 compiler: gcc v: 7.5.0 
           Desktop: Xfce 4.14.1 tk: Gtk 3.22.30 wm: xfwm4 dm: LightDM 
           Distro: Linux Mint 19.3 Tricia base: Ubuntu 18.04 bionic 
Machine:   Type: Laptop System: LENOVO product: 0769BSG v: 3000 N200 serial: <filter> 
           Chassis: No Enclosure type: 10 serial: <filter> 
           Mobo: LENOVO model: IEL10 v: REFERENCE serial: <filter> BIOS: LENOVO v: 68ET37WW 
           date: 06/11/2008 
Battery:   ID-1: BAT1 charge: 53.8 Wh condition: 56.6/56.6 Wh (100%) volts: 11.1/11.1 
           model: COMPAL PA3465U serial: <filter> status: Charging 
           Device-1: hidpp_battery_0 model: Logitech Wireless Mouse M187 serial: <filter> 
           charge: 55% status: Discharging 
CPU:       Topology: Dual Core model: Intel Core2 Duo T7250 bits: 64 type: MCP arch: Core Merom 
           rev: D L2 cache: 2048 KiB 
           flags: lm nx pae sse sse2 sse3 ssse3 vmx bogomips: 7980 
           Speed: 2168 MHz min/max: 800/2001 MHz Core speeds (MHz): 1: 2153 2: 1996 
Graphics:  Device-1: NVIDIA G72M [Quadro NVS 110M/GeForce Go 7300] vendor: Lenovo driver: nouveau 
           v: kernel bus ID: 01:00.0 chip ID: 10de:01d7 
           Display: x11 server: X.Org 1.20.8 driver: modesetting,nouveau unloaded: fbdev,vesa 
           resolution: 1280x800~60Hz 
           OpenGL: renderer: NV46 v: 2.1 Mesa 20.0.8 direct render: Yes 
Audio:     Device-1: Intel 82801H HD Audio vendor: Lenovo driver: snd_hda_intel v: kernel 
           bus ID: 00:1b.0 chip ID: 8086:284b 
           Sound Server: ALSA v: k5.4.0-65-generic 
Network:   Device-1: Intel PRO/Wireless 4965 AG or AGN [Kedron] Network driver: iwl4965 
           v: in-tree: port: 1c00 bus ID: 04:00.0 chip ID: 8086:4230 
           IF: wlp4s0 state: up mac: <filter> 
           Device-2: Broadcom and subsidiaries NetLink BCM5906M Fast Ethernet PCI Express 
           vendor: Lenovo driver: tg3 v: 3.137 port: 1c00 bus ID: 06:00.0 chip ID: 14e4:1713 
           IF: enp6s0 state: down mac: <filter> 
Drives:    Local Storage: total: 232.89 GiB used: 101.48 GiB (43.6%) 
           ID-1: /dev/sda vendor: Samsung model: SSD 840 EVO 250GB size: 232.89 GiB 
           speed: 3.0 Gb/s serial: <filter> 
Partition: ID-1: / size: 228.23 GiB used: 101.48 GiB (44.5%) fs: ext4 dev: /dev/sda1 
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 4 rev: 2.0 chip ID: 1d6b:0002 
           Device-1: 1-4:3 info: Chicony USB 2.0 Camera type: Video driver: uvcvideo rev: 2.0 
           chip ID: 04f2:b013 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 6 rev: 2.0 chip ID: 1d6b:0002 
           Hub: 3-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
           Hub: 4-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
           Device-2: 4-1:2 info: AuthenTec AES2501 Fingerprint Sensor type: <vendor specific> 
           driver: N/A rev: 1.1 chip ID: 08ff:2580 
           Hub: 5-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
           Hub: 6-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
           Device-3: 6-1:2 info: Logitech Unifying Receiver type: Mouse,HID 
           driver: logitech-djreceiver,usbhid rev: 2.0 chip ID: 046d:c52f 
           Device-4: 6-2:3 info: Broadcom BCM2045 Bluetooth type: Bluetooth driver: btusb rev: 2.0 
           chip ID: 0a5c:2101 
           Hub: 7-0:1 info: Full speed (or root) Hub ports: 2 rev: 1.1 chip ID: 1d6b:0001 
Sensors:   System Temperatures: cpu: 45.0 C mobo: N/A gpu: nouveau temp: 33 C 
           Fan Speeds (RPM): N/A 
Repos:     No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb http: //mirror6.layerjet.com/linuxmint/packages tricia main upstream import backport
           2: deb http: //archive.ubuntu.com/ubuntu bionic main restricted universe multiverse
           3: deb http: //archive.ubuntu.com/ubuntu bionic-updates main restricted universe multiverse
           4: deb http: //archive.ubuntu.com/ubuntu bionic-backports main restricted universe multiverse
           5: deb http: //security.ubuntu.com/ubuntu/ bionic-security main restricted universe multiverse
           6: deb http: //archive.canonical.com/ubuntu/ bionic partner
Info:      Processes: 187 Uptime: 1m Memory: 3.84 GiB used: 1.04 GiB (27.1%) Init: systemd v: 237 
           runlevel: 5 Compilers: gcc: 7.5.0 alt: 7 Client: Unknown python3.6 client inxi: 3.0.32 
[/code]

file 20.xorg.conf

Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

Barabazs, 2021-02-07:
Please clarify what problem you are having in a more detailed and accurate manner.

goyon2000, 2021-02-08:
Ok. Hope my english will not be too bad.

Once I connect the usb, the system becomes very slow and unresponsive. It is very difficult to adjust my two monitor as i can barely use the mouse and even the keyboard. I try to use the command xrandr auto and others xrandr off. I try also to put one monitor to the right of the other. It worked once but even like that it was still very slow. it is like the three screen overlap (laptop and the two other one). 

I can see that in the /var/log/displaylink/DisplayLinkManager.log there is a lot of strange characters. I can send the file if you want.

goyon2000, 2021-02-09:
Sorry I think I closed the issue but for me this is still open

Barabazs, 2021-02-09:
Is your laptop lid open or closed?

goyon2000, 2021-02-10:
It was opened

Barabazs, 2021-02-11:
The issue your describing is unrelated to `displaylink-debian`. 
It's probably your hardware that does not meet the minimum requirements.

closed 2021-02-11 by Barabazs