AdnanHodzic/displaylink-debian #537: ThinkPad T945 and ThinkPad Hybrid USB-A/C dock issue

EmadMokhtar, 2021-02-15:
# Overview

I have ThinkPad T945 and ThinkPad Hybrid USB-A/C dock and for one month now I'm trying to fix a laggy issue with my external monitor. I saw your package and Installed it, After the reboot, my monitors are not detected. 

I tried these configurations with no luck (laggy display)

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

## Debug Info

--------------- Linux system info ----------------

Distro: Pop
Release: focal
Kernel: 5.8.0-7642-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0+dfsg
1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev d1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x57 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:06:00.0

Barabazs, 2021-02-15:
Hi @EmadMokhtar Are you using the proprietary amdgpu-pro driver or the open source one?

EmadMokhtar, 2021-02-15:
I'mnot sure. I'm using the one comes with PopOS installation

Barabazs, 2021-02-16:
Well you should try to find out, as it might be related to https://displaylink.org/forum/showthread.php?t=67378

EmadMokhtar, 2021-02-16:
> Well you should try to find out, as it might be related to https://displaylink.org/forum/showthread.php?t=67378

I will give it a try and let you know.

closed 2021-03-01 by Barabazs