AdnanHodzic/displaylink-debian #538: Ubuntu 20.10 evdi install issue kernel config

ascju, 2021-02-16:
Hi there.

I am not getting my ASUS MB168B to work with Ubuntu.
When trying to install, I run into the following issue when the setup is trying to install evdi:

--------------
Cloning into 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 8 (delta 0), pack-reused 0
Unpacking objects: 100% (46/46), 67.99 KiB | 273.00 KiB/s, done.
Note: switching to '5bd169208b8dbce6ab3c63643bb69eba79594276'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1]: Entering directory '/home/juc/Repositories/displaylink-debian/evdi/module'
cat: /etc/redhat-release: No such file or directory
make -C /lib/modules/5.8.0-43-generic/build M=$PWD
make[2]: Entering directory '/usr/src/linux-headers-5.8.0-43-generic'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[2]: *** [Makefile:746: include/config/auto.conf] Error 1
make[2]: Leaving directory '/usr/src/linux-headers-5.8.0-43-generic'
make[1]: *** [Makefile:77: module] Error 2
make[1]: Leaving directory '/home/juc/Repositories/displaylink-debian/evdi/module'
make: *** [Makefile:8: all] Error 2
cp: cannot stat 'library/libevdi.so.1.*': No such file or directory

Any idea what can be done here?

Here is my system info in case this helps:

-------------- Linux system info ----------------

Distro: Ubuntu
Release: groovy
Kernel: 5.8.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Iris Plus Graphics 640 (rev 06)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Thanks!

Barabazs, 2021-02-20:
Please try to download the repo as a zip instead of cloning the repo.

ascju, 2021-02-20:
Hi Barabazs!

Thanks for you input - unfortunately no luck.

I downloaded the ZIP, unzip the package, and run the setup (with sudo). 
Unfortunately it is giving the exact same output as postet above - so it behaves exactly like the clone from git.

Any other ideas what could help? Thanks!

ascju, 2021-02-23:
Quick note - maybe this helps to identify the cause:

I tried installing the official displaylink driver (5.3.1.34) and it gives me this:

------

Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  
DisplayLink Linux Software 5.3.1.34 install script called: 
Distribution discovered: Ubuntu 20.10

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.7.0/source ->
                 /usr/src/evdi-1.7.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.8.0-43-generic all INCLUDEDIR=/lib/modules/5.8.0-43-generic/build/include KVERSION=5.8.0-43-generic DKMS_BUILD=1...(bad exit status: 2)
ERROR (dkms apport): binary package for evdi: 1.7.0 not found
Error! Bad return status for module build on kernel: 5.8.0-43-generic (x86_64)
Consult /var/lib/dkms/evdi/1.7.0/build/make.log for more information.

Barabazs, 2021-02-23:
I made a small adjustment on my fork. Can you try with this one please.
https://github.com/Barabazs/displaylink-debian

ascju, 2021-02-23:
Thanks for your effort. Really appreciated!
Unfortunately no change.
I tried both, git clone as well as download the zip.
It is giving me the exact same error as mentioned above.

Barabazs, 2021-02-24:
Can you try to down- or upgrade your kernel? Reboot with another kernel version?

ascju, 2021-02-26:
Hey! I updated to 

--------------- Linux system info ----------------

Distro: Ubuntu
Release: groovy
Kernel: 5.8.0-44-generic

which is only a minor change 5.8.0-43 but Igave it a shot - still the same.

As this is my work laptop, I want to limit the attempts... :) 
Can you suggest a Kernel version which I should try?

Barabazs, 2021-02-27:
Can you post the output of:
 `sudo dkms status evdi`
`systemctl status displaylink-driver`
`sudo displaylink-debian.sh --debug`

ascju, 2021-02-28:
Sure:

$ sudo dkms status evdi
evdi, 1.7.0: added
evdi, 1.7.2: added
evdi, 5.2.14: added

-----

$ systemctl status displaylink-driver
Unit displaylink-driver.service could not be found.

------

Debug still gives the same as above as far as I can tell but anyways, here it it:

$ sudo displaylink-debian.sh --debug
...
--------------- Linux system info ----------------

Distro: Ubuntu
Release: groovy
Kernel: 5.8.0-44-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.7.2
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Iris Plus Graphics 640 (rev 06)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Barabazs, 2021-03-01:
Do you have build-essential installed?

ascju, 2021-03-01:
yes. current version.

$ dpkg -s build-essential

Package: build-essential
Status: install ok installed
Priority: optional
Section: devel
Installed-Size: 21
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Version: 12.8ubuntu3
Depends: libc6-dev | libc-dev, gcc (>= 4:9.2), g++ (>= 4:9.2), make, dpkg-dev (>= 1.17.11)

Barabazs, 2021-03-06:
It's clearly unrelated to displaylink-debian, so there is nothing we can do for you.
I'll leave it open for a while. Maybe someone else knows how to resolve your issue.

Barabazs, 2021-03-07:
This might be relevant for you https://github.com/DisplayLink/evdi/issues/262#issuecomment-792280725

ascju, 2021-03-07:
will give it a try and give an update. 
thanks for your support!

closed 2021-03-13 by Barabazs