AdnanHodzic/displaylink-debian #539: displaylink-debian.sh doesnt do anything

thedenbrews, 2021-02-17:
When I execute displaylink-debian.sh with sudo I get only this output.

No protocol specified
Can't open display :0

I am running the latest version of Debian 10 with everything standard.  Its not installing so I cannot send any logs even.  Not sure what is going on.

Barabazs, 2021-02-17:
What command do you run exactly?

thedenbrews, 2021-02-17:
sudo ./displaylink-debian.sh 
While in the directory downloaded by git.

This is all I saw on your instructions.

Barabazs, 2021-02-17:
Are you running Wayland or Xorg/X11/...?

thedenbrews, 2021-02-17:
I think it’s xorg as it’s the standard Debian 10 desktop.

thedenbrews, 2021-02-18:
Oh, I see Xwayland in top so I am assuming now it is Wayland.

Barabazs, 2021-02-23:
Is your display variable set?
```
echo $DISPLAY
echo $XAUTHORITY
```

thedenbrews, 2021-02-24:
> 
> 
> Is your display variable set?
> 
> echo $DISPLAY
:0
> echo $XAUTHORITY
<blank>

Barabazs, 2021-02-25:
Please post the output of `sudo ./displaylink-debian.sh --debug`

thedenbrews, 2021-02-25:
caleb@DELLXPS13:~/displaylink-debian$ sudo ./displaylink-debian.sh --debug
[sudo] password for caleb:
No protocol specified
Can't open display :0
caleb@DELLXPS13:~/displaylink-debian

Barabazs, 2021-02-27:
You're not trying to do this over ssh right?

thedenbrews, 2021-02-28:
No, but I removed Debian and install ubuntu so now it works, but it is puzzling.  Sorry to waste your time on this.

closed 2021-02-28 by Barabazs

superleo79, 2021-03-04:
Hello, I reopen the issue because I am having the same problem.

I am running Ubuntu 18.04. I will paste here some information about my system

uname -a gives this line:

Linux ldc1-Lenovo-ideapad-320S-15IKB 5.4.0-66-generic #74~18.04.2-Ubuntu SMP Fri Feb 5 11:17:31 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

lspci | grep VGA this the following graphic card:

00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)

It is an Intel one, so following this official Displaylink support post [https://support.displaylink.com/knowledgebase/articles/1181623-displaylink-ubuntu-driver-after-recent-x-upgrades](url) I created 
a file 20-displaylink.conf under  /usr/share/X11/xorg.conf.d/

with the following content: 

Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection  

I must say that I am able to install the official driver displaylink-driver-5.1.26.run and the terminal says that it is installed, but no monitor appears under setting/displays 

Due to that fact, I found on internet your driver but I am having the same problems than the OP had.

When I run sudo ./displaylink-debian.sh  I get:

No protocol specified
Can't open display :0

When I run sudo ./displaylink-debian.sh I get the same answer:

No protocol specified
Can't open display :0

After running xrandr --listproviders I get:

Providers: number : 0
 and anything else.

I tried to run the divice on Windows to check if it works and it is ok there. How can I make it run on Ubuntu?

superleo79, 2021-03-05:
Problem solved¡¡¡¡

Maybe it is good that to code a check to avoid problems like mine.

The problem was that I had to disable WaylandEnable in order to use Xorg by defaul when booting the system.

I had to edit as sudo this file /etc/gdm3/custom.conf and  change line FROM:
#WaylandEnable=false
TO:
WaylandEnable=false
By uncommenting the above line your system will use the Xorg display manager instead of Wayland next time it boots.

I got this info form the folowwing link [https://linuxconfig.org/how-to-disable-wayland-and-enable-xorg-display-server-on-ubuntu-18-04-bionic-beaver-linux](url)