AdnanHodzic/displaylink-debian #543: Please set an environment variable SYSTEMINITDAEMON

unnome, 2021-02-19:
Hellos! I can't seem to be able to proceed with the installation as I am requested to set an environment variable.

The symptoms are very similar to [issue 445](https://github.com/AdnanHodzic/displaylink-debian/issues/445)

```
Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
ERROR: the installer script is unable to find out how to start DisplayLinkManager service automatically on your system.
Please set an environment variable SYSTEMINITDAEMON to 'upstart', 'systemd' or 'runit' before running the installation script to force one of the options.
Installation terminated.
```

This is my debug output
```
$ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-6-amd64

---------------- DisplayLink info ----------------

Driver version: 
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: Ltd.
VGA: NVIDIA Corporation GM200 [GeForce GTX 980 Ti] (rev a1)
VGA (3D): 
X11 version: 1.20.4-1+deb10u2
X11 configs: /etc/X11/xorg.conf.d/synaptics.conf /etc/X11/xorg.conf.d/10-evdev.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x1b9 cap: 0x1, Source Output crtcs: 4 outputs: 9 associated providers: 0 name:NVIDIA-0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

- Setting the variable doesn't seem to help. I tried all 3 suggested values, to no avail.
- Exporting the variable doesn't seem to help. I tried all 3 suggested values, to no avail.

Any help would be appreciated :)

Thanks!

Barabazs, 2021-02-23:
How are you running debian?

unnome, 2021-02-23:
Thanks for your reply! I'm using Debian in a MX Linux 19.2

Barabazs, 2021-02-24:
Apparently systemd is disabled by default on your distro.You should enable it and then retry the installation: https://mxlinux.org/wiki/system/systemd/#enablingsystemd

unnome, 2021-02-24:
thank you! I tried running with systemd however the boot failed, complaining that it "failed to start Load AppArmor". I will note that the error log had several mentions of that failure message. 

I then installed AppArmor, however am not yet able to run debian with systemd. The only visible effect is that the "failed to start Load AppArmor" appears only once in the error log.

I know that my comment is not very useful, but I'm eager to follow your instructions

Barabazs, 2021-02-25:
Sorry, I can't help you with that. You'll have to search on your distribution website or forum.

unnome, 2021-02-25:
It makes sense. Thanks Barabazs. I'll close this ticket and open a new one if restarting with systemd enable doesn't solve the issue.

closed 2021-02-25 by unnome