AdnanHodzic/displaylink-debian #545: [HELP] Error kernel configuration is invalid

danygit, 2021-02-24:
Hello all, (sorry for my english).
I restart my computer yesterday and now my monitors are not detected.
I try many thing and also displaylink-debian and I have this error : 

`./evdi.sh: ligne 10 : [: trop d'arguments
Clonage dans 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 8 (delta 0), pack-reused 0
Dépaquetage des objets: 100% (46/46), 67.99 Kio | 275.00 Kio/s, fait.
Note : basculement sur '5bd169208b8dbce6ab3c63643bb69eba79594276'.

Vous êtes dans l'état « HEAD détachée ». Vous pouvez visiter, faire des modifications
expérimentales et les valider. Il vous suffit de faire un autre basculement pour
abandonner les commits que vous faites dans cet état sans impacter les autres branches

Si vous voulez créer une nouvelle branche pour conserver les commits que vous créez,
il vous suffit d'utiliser l'option -c de la commande switch comme ceci :

  git switch -c <nom-de-la-nouvelle-branche>

Ou annuler cette opération avec :

  git switch -

Désactivez ce conseil en renseignant la variable de configuration advice.detachedHead à false

CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1] : on entre dans le répertoire « /home/dropital/Téléchargements/DisplayLink USB Graphics Software for Ubuntu 5.3.1/displaylink-driver-5.3.1.34/displaylink-debian/evdi/module »
cat: /etc/redhat-release: Aucun fichier ou dossier de ce type
make -C /lib/modules/5.8.0-43-generic/build M=$PWD
make[2] : on entre dans le répertoire « /usr/src/linux-headers-5.8.0-43-generic »

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[2]: *** [Makefile:746 : include/config/auto.conf] Erreur 1
make[2] : on quitte le répertoire « /usr/src/linux-headers-5.8.0-43-generic »
make[1]: *** [Makefile:77 : module] Erreur 2
make[1] : on quitte le répertoire « /...ts/DisplayLink USB Graphics Software for Ubuntu 5.3.1/displaylink-driver-5.3.1.34/displaylink-debian/evdi/module »
make: *** [Makefile:8 : all] Erreur 2
cp: impossible d'évaluer 'library/libevdi.so.1.*': Aucun fichier ou dossier de ce type
`

And her the log : 
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.8.0-43-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.7.2
5.1.26
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

danygit, 2021-02-24:
I have tried with only install displaylink and also with this repo.

`Checking dependencies

unzip is installed
linux-headers-5.8.0-43-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: Aucun fichier ou dossier de ce type

-------------------------------------------------------------------

Performing clean-up

`

danygit, 2021-02-24:
`evdi, 1.7.0: added
evdi, 1.7.2: added
evdi, 5.1.26: added
evdi, 5.2.14: added
`

danygit, 2021-02-24:
Thanks to help me ;)

Barabazs, 2021-02-24:
No problem.
What does` systemctl status displaylink`give? 
You should try to remove/uninstall all those evdi entries.
`sudo dkms remove evdi/1.7.0 - - all` and for the next one, replace 1.7.0 with 1.7.2 and so on. Also reboot when prompted and before retrying `sudo dkms status evdi`

danygit, 2021-02-24:
`Unit displaylink.service could not be found.`

`sudo dkms status evdi
evdi, 5.1.26: added
evdi, 5.2.14: added
`

danygit, 2021-02-24:
I should remove 5.1.26 too ? and 5.2.14 ?

Barabazs, 2021-02-24:
Yes all of them.

danygit, 2021-02-24:
I have delete all and reboot.
After reboot, i check and no evdi installed.
I clone again this repo.
I launch again and i have the same problem

`Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  

-------------------------------------------------------------------
Clonage dans 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 8 (delta 0), pack-reused 0
Dépaquetage des objets: 100% (46/46), 67.99 Kio | 819.00 Kio/s, fait.
Note : basculement sur '5bd169208b8dbce6ab3c63643bb69eba79594276'.

Vous êtes dans l'état « HEAD détachée ». Vous pouvez visiter, faire des modifications
expérimentales et les valider. Il vous suffit de faire un autre basculement pour
abandonner les commits que vous faites dans cet état sans impacter les autres branches

Si vous voulez créer une nouvelle branche pour conserver les commits que vous créez,
il vous suffit d'utiliser l'option -c de la commande switch comme ceci :

  git switch -c <nom-de-la-nouvelle-branche>

Ou annuler cette opération avec :

  git switch -

Désactivez ce conseil en renseignant la variable de configuration advice.detachedHead à false

CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1] : on entre dans le répertoire « /h---/displaylink-debian/evdi/module »
cat: /etc/redhat-release: Aucun fichier ou dossier de ce type
make -C /lib/modules/5.8.0-43-generic/build M=$PWD
make[2] : on entre dans le répertoire « /usr/src/linux-headers-5.8.0-43-generic »

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[2]: *** [Makefile:746 : include/config/auto.conf] Erreur 1
make[2] : on quitte le répertoire « /usr/src/linux-headers-5.8.0-43-generic »
make[1]: *** [Makefile:77 : module] Erreur 2
make[1] : on quitte le répertoire « /ho----s/displaylink-debian/evdi/module »
make: *** [Makefile:8 : all] Erreur 2
cp: impossible d'évaluer 'library/libevdi.so.1.*': Aucun fichier ou dossier de ce type
`

danygit, 2021-02-24:
Now I have :

systemctl status displaylink-driver
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/lib/systemd/system/displaylink-driver.service; static; vendor preset: enabled)
     Active: inactive (dead)

Barabazs, 2021-02-24:
Can you try to down- or upgrade your kernel? Reboot with another kernel version?

danygit, 2021-02-24:
I upgrade today. I cant reboot to an another kernel version.

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"

danygit, 2021-02-25:
Hello, do you have an other idea ? I really nead displaylink and my company too.

danygit, 2021-02-26:
I will try to reset this week and retry all step. if you have an idea in the same time, ;)

Barabazs, 2021-02-27:
Do you have build-essential installed?

closed 2021-03-06 by Barabazs