AdnanHodzic/displaylink-debian #546: Had display link working but now not so much

smalco, 2021-02-25:
Had displaylink working on Linux Mint (still haven't figured it out on Kali) however I just randomly stopped working. Tried to reinstall and Debug but all I get is. USB screen preloads but then black. Not sure what happened. Any help would be appreciated thanks.
<pre>
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyssa
Kernel: 5.4.0-66-generic

---------------- DisplayLink info ----------------

Driver version: 1.3.54
1.4.210
1.7.0
1.7.0+dfsg
1.7.2
4.2.29
5.2.14
DisplayLink service status: 
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Haswell-ULT
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x120 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xfe cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xdc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xba cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
</pre>

Barabazs, 2021-02-25:
Copy this config to /etc/X11/xorg.conf.d/20-displaylink.conf, then reboot and repost the debug output please.
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

smalco, 2021-02-26:
Sorry for the delay I have had with my other monitor at work. So went to
install that patch and I make it to xorg.conf.d and there are no sub
folders or prompts inside the folded not even in root with the show hidden
folders on there is no 20-displaylink.conf

-- 
Malcolm, Stephen
SSG. INARNG

 Maintenance Manager NCO
NGIN-IBL-A
HHC 113 BSB
401 N. Country Club Rd.
Muncie, IN. 47303
Cell: 317-938-7769

"It has been my observation that most people get ahead during the time that
others waste.”

- Henry Ford

Barabazs, 2021-02-27:
I'm not sure I understand what you are saying.

Did you copy the config to the file?

closed 2021-03-06 by Barabazs