AdnanHodzic/displaylink-debian #548: Issue with displaylink second screen on Parrot OS

RogierKonings, 2021-03-01:
Dear,

drivers seem to install properly, but my second screen connected to the displaylink adapter does not seem to work.

I am running Parrot OS (Debian) and have a displaylink adapter with two screens (1 HDMI, 1 DP)

This is my debug output:

--------------- Linux system info ----------------

Distro: Parrot
Release: n/a
Kernel: 5.10.0-3parrot1-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: 
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.10-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0x153 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x131 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x10f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xed cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Any idea what the issue is?

Barabazs, 2021-03-01:
You need to associate the providers:
[post install guide/display-detection](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection)

RogierKonings, 2021-03-01:
problem still exists (had tried this also earlier but restarted)

--------------- Linux system info ----------------

Distro: Parrot
Release: n/a
Kernel: 5.10.0-3parrot1-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: 
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.10-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x153 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x131 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x10f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xed cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

Barabazs, 2021-03-01:
As far as displaylink-debian is concerned, everything looks OK.

Can you be more specific about the problem you are experiencing?

What is the output of `xrandr --listmonitors`? (be sure to have the providers associated first)

RogierKonings, 2021-03-01:
Monitors: 2
 0: +*DP-1 1920/597x1080/336+1920+0  DP-1
 1: +eDP-1 1920/309x1080/173+0+0  eDP-1

I have two external monitors connected to my displaylink adapter: one via DisplayPort and the other one via HDMI. 

When i connect the displaylink adapter, I can only see one of my connected screens: the HDMI screen.

I want to have both screens working: it works on macOS, and also on Ubuntu version, so hardware wise everything should be in order.

Barabazs, 2021-03-06:
At this point it comes down to tweaking your settings.

closed 2021-03-06 by Barabazs