AdnanHodzic/displaylink-debian #549: ubuntu groovy black screen problem even with monitor is detected

khoubeibba, 2021-03-01:
Hello everyone o/

I tried everything, installation seems to be OK : 

Ubuntu 20.10 
```
#lspci -vnn | grep VGA -A 12                                                                                                                                                                                    
00:02.0 VGA compatible controller [0300]: Intel Corporation Iris Plus Graphics G7 [8086:8a52] (rev 07) (prog-if 00 [VGA controller])
	DeviceName: To Be Filled by O.E.M.
	Subsystem: Dell Iris Plus Graphics G7 [1028:097c]
	Flags: bus master, fast devsel, latency 0, IRQ 139
	Memory at b0000000 (64-bit, non-prefetchable) [size=16M]
	Memory at a0000000 (64-bit, prefetchable) [size=256M]
	I/O ports at 5000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
	Capabilities: [d0] Power Management version 2
	Capabilities: [100] Process Address Space ID (PASID)

#lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
i915
```

-------------------------------
`dkms status evdi                                                                                                                                                                                                ✔ evdi, 1.7.2, 5.8.0-44-generic, x86_64: installed`
-------------------------------
```
cat /etc/X11/xorg.conf.d/20-displaylink.conf                                                                                                                                                                         
Section "Device"
	Identifier  "DisplayLink"
	Driver      "modesetting"
	Option      "PageFlip" "false"
EndSection
```
---------------------------------
```
xrandr --listproviders                                                                                                                                                                                               
Providers: number : 5
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0x147 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x125 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x103 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```
 
I tried to force the intel card in /etc/X11/xorg.conf.d/20-displaylink.conf  with 
```
Section "Device"
	 Identifier "Intel"
         Driver "intel"
EndSection

```
it will detect only one provider : intel without the modesetstring for the displaylink 

when unplug the device : 
```
Mar  1 12:54:42 dell-inspiron kernel: [ 4681.702623] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
Mar  1 12:54:42 dell-inspiron kernel: [ 4681.702631] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
Mar  1 12:54:42 dell-inspiron kernel: [ 4681.702637] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
Mar  1 12:54:42 dell-inspiron kernel: [ 4681.702642] evdi: [D] evdi_detect:94 (dev=-1) poll connector state: disconnected
Mar  1 12:54:42 dell-inspiron /usr/libexec/gdm-x-session[48015]: (II) modeset(0): EDID vendor "CMN", prod id 5605
Mar  1 12:54:42 dell-inspiron /usr/libexec/gdm-x-session[48015]: (II) modeset(0): Printing DDC gathered Modelines:
Mar  1 12:54:42 dell-inspiron /usr/libexec/gdm-x-session[48015]: (II) modeset(0): Modeline "1920x1080"x0.0  152.84  1920 2000 2054 2250  1080 1086 1094 1132 +hsync -vsync (67.9 kHz eP)
Mar  1 12:54:42 dell-inspiron /usr/libexec/gdm-x-session[48015]: (II) modeset(0): Modeline "1920x1080"x0.0  122.27  1920 2000 2054 2250  1080 1086 1094 1132 +hsync -vsync (54.3 kHz e)
Mar  1 12:54:42 dell-inspiron /usr/libexec/gdm-x-session[48015]: (II) modeset(0): EDID vendor "CMN", prod id 5605
Mar  1 12:54:42 dell-inspiron /usr/libexec/gdm-x-session[48015]: (II) modeset(0): Printing DDC gathered Modelines:
Mar  1 12:54:42 dell-inspiron /usr/libexec/gdm-x-session[48015]: (II) modeset(0): Modeline "1920x1080"x0.0  152.84  1920 2000 2054 2250  1080 1086 1094 1132 +hsync -vsync (67.9 kHz eP)
[... 105 lines cut ...]
Mar  1 12:55:57 dell-inspiron kernel: [ 4757.103399] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
Mar  1 12:55:57 dell-inspiron kernel: [ 4757.103407] evdi: [D] evdi_detect:90 (dev=0) poll connector state: connected
Mar  1 12:55:57 dell-inspiron kernel: [ 4757.103409] evdi: [D] evdi_painter_get_edid_copy:228 (dev=0) EDID valid
Mar  1 12:55:57 dell-inspiron rtkit-daemon[3515]: Supervising 4 threads of 2 processes of 1 users.
Mar  1 12:55:57 dell-inspiron rtkit-daemon[3515]: Successfully made thread 55429 of process 47926 owned by '1000' RT at priority 5.
Mar  1 12:55:57 dell-inspiron rtkit-daemon[3515]: Supervising 5 threads of 2 processes of 1 users.
Mar  1 12:55:57 dell-inspiron gsd-media-keys[48394]: Unable to get default sink
Mar  1 12:55:57 dell-inspiron colord[4214]: failed to get session [pid 48388]: Aucune donnée disponible
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Error while loading thumbnail: Error: Argument width: value is out of range for gfloat
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: message repeated 9 times: [ Error while loading thumbnail: Error: Argument width: value is out of range for gfloat]
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 38 with keysym 38 (keycode 11).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 39 with keysym 39 (keycode 12).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 35 with keysym 35 (keycode e).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 36 with keysym 36 (keycode f).
Mar  1 12:55:58 dell-inspiron gnome-shell[48258]: Window manager warning: Overwriting existing binding of keysym 37 with keysym 37 (keycode 10).
Mar  1 12:55:58 dell-inspiron fwupd[4570]: 11:55:58:0336 FuEngine             solokey failed to change udev device /sys/devices/platform/evdi.0/drm/card0: cannot ensure ID: FuUdevDevice:
Mar  1 12:55:58 dell-inspiron fwupd[4570]: Unknown Device
Mar  1 12:55:58 dell-inspiron fwupd[4570]:   Flags:                none
```

Any help please :)

khoubeibba, 2021-03-01:
[image]

Barabazs, 2021-03-01:
Please post the output of `sudo ./displaylink-debian.sh --debug`

khoubeibba, 2021-03-01:
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: groovy
Kernel: 5.8.0-44-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Iris Plus Graphics G7 (rev 07)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX230] (rev a1)
X11 version: 1.20.9-2ubuntu1.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
#	Identifier "Intel"
#	Driver "intel"

	Identifier  "DisplayLink"
	Driver      "modesetting"
	Option      "PageFlip" "false"

#	Option "AccelMethod" "uxa"
#	Option "VSync" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0x147 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x125 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x103 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

khoubeibba, 2021-03-01:
Any clue please ?? I passed all the day trying without any progress, the screen is detected and remains black

Barabazs, 2021-03-06:
Please try with this config:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

khoubeibba, 2021-03-07:
Hello @Barabazs  thanks for your help, same thing : black screen

Barabazs, 2021-03-08:
You'll have to finetune your config and xrandr commands. The screen is detected, so displaylink does work.

closed 2021-03-08 by Barabazs