AdnanHodzic/displaylink-debian #552: Can't make it work on latest kali release

sleepdisastr, 2021-03-15:
Hi, 

Im experiencing some issue trying to install dp on the latest kali rolling version. Im getting the following error while running the script 
Reading state information... Done
E: Unable to locate package linux-headers-5.10.0-kali3-amd64
E: Couldn't find any package by glob 'linux-headers-5.10.0-kali3-amd64'
E: Couldn't find any package by regex 'linux-headers-5.10.0-kali3-amd64'
linux-headers-5.10.0-kali3-amd64 installation failed.  Aborting.

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 5.10.0-kali3-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [17aa:507c]
VGA: Intel Corporation CometLake-U GT2 [UHD Graphics] (rev 02)
VGA (3D): 
X11 version: 1.20.10-3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

Would you be so kind to add support for the latest rolling kali version ? 

Thanks in advance. 
S.

sldbs, 2021-03-15:
solved somehow

closed 2021-03-15 by sldbs

tristan-k, 2021-06-06:
I'm having the exact same issue with Proxmox 6 based on Debian GNU/Linux 10 (buster) x86_64. How did you solve the issue?