AdnanHodzic/displaylink-debian #553: Support for Raspbian

jkimsis, 2021-03-24:
Hello,

I want to install displaylink-debian on a Raspberry Pi 400. It didn't work out out of the box because the kernel files have a Raspbian specific naming, but I managed to modify the script to accept the kernel files. However, installing EVDI fails (errors in the attached file). I would like to contribute to your project by getting it to work on Raspbian, but need some help.

Attached modified debug output and errors when installing, modified displaylink-debian in pastebin link
https://pastebin.com/6M6XCe40
[install_error.txt](https://github.com/AdnanHodzic/displaylink-debian/files/6196010/install_error.txt)
[debug_raspbian.txt](https://github.com/AdnanHodzic/displaylink-debian/files/6195944/debug_raspbian.txt)

Barabazs, 2021-03-24:
You will need ARM binaries for displaylink and probably a specific evdi version that is compatible with that displaylink version.

jkimsis, 2021-03-24:
The problem appears to be that evdi installer can't find a specific header:
/var/lib/dkms/evdi/1.6.0+dfsg/build/evdi_modeset.c:15:10: fatal error: drm/drmP.h: No such file or directory
 #include <drm/drmP.h>

I can't mind any such header in my system.

closed 2021-03-24 by jkimsis