AdnanHodzic/displaylink-debian #555: Ubuntu 18.04.5 - Black screen and only mouse with intel/modesettings driver combination

nightsparc, 2021-03-24:
Hi guys!

First of all, amazing work (and support of course)!

As stated in the title, I'm getting the dreaded black screen / mouse behavior as described (and handled?) in #68, #293, #429

I first thought #293 was similar to my problem here, however **rrajkomars** setup was finally solved by @Barabazs by ensuring that the provider 0 (the actual graphics provider) is using the `modesettings` driver instead the `intel` driver. Actually, this setup _does_ work for me after some tweaking (basically following the tips at the end of #429 and/or [the DL Ubuntu troubleshooting page](https://support.displaylink.com/knowledgebase/articles/1181623-displaylink-ubuntu-driver-after-recent-x-upgrades).

```sh
Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection
```
and 

```sh
Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0xf9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xd7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xb5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x7a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

Anyway, I fear, my setup is a bit different. I must be able to disable triple buffering and use some kind of Hw acceleration on my primary display (connected to the Intel GPU), which both are not really working in the modesettings driver. In addition, due to a lack of ports, I have to 

The setup using the `modesettings`-driver is working (at least most of the time, sometimes it just dies...)

I'm working on a ruggedized piece of horrible hardware, which is connected to another horrible piece of hardware. So, I'm having a full HD display which is only capable of running at 30 fps (don't ask...), and need to connect to another display via DisplayLink.

Thus, I think the primary display should be using the intel driver, and the DL should be running on the modesettings driver. However, that setup just doesn't work.

## debug output

```sh

-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.4.0-70-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: up and running
EVDI service version: 1.7.2

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Integrated
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2~18.04.4
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/10-intel.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection  

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 4 name:Intel
Provider 1: id: 0x10b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xe9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xc7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x8c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

And the `10-intel.conf`

```sh
Section "Device"
  Identifier  "Intel"
  Driver      "intel"
  # only supports 30Hz rendering
  Option      "TripleBuffer" "off"  
  Option      "VSync" "false"  
EndSection
```

As I said before, I must be able to run the DL adapter besides the internal GPU and I'm also in the need of some hardware acceleration - when using the `modesettings` driver, my applications are running at load of about ~98-100% of the CPU - **without** having _anything_ rendered on the display connected via DL. If my applications are running on the same system _without_ the `modesettings` driver, I hit around 60-70%...

Any idea how to get both drivers up and running?

Thanks in advance for any idea :)

Barabazs, 2021-05-15:
@nightsparc did you get it to work as you wanted?

nightsparc, 2021-05-15:
@Barabazs  Yes, no, in parts :). I was ending up with a dual-head configuration, with the primary display attached to the first xorg-server using the `intel` driver and the second display/xorg using the `modesettings` driver, but using llvmpipe/Gallium for rendering and thus causing a high CPU load...

Ok, that solved our first problem - tripple-buffering was working again and the primary display was running on the intel iGPU. However, the combined loads of both displays/xorg-servers were just too much for our desired setup. The Gallium-rendered display caused an increase of the CPU load of about 20-30%, still running nothing besides Xfce. Which means that the system was running with a load of about 90%. So, we basically ended up using a 2nd computer :see_no_evil:

closed 2021-06-03 by Barabazs