AdnanHodzic/displaylink-debian #559: New Pop OS! support

ghahramani, 2021-04-02:
Hi, 

I tried to install it as you instructed but unfortunately, it shows compile error when installing evdi on the os version below:

```
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 20.10
Release:	20.10
Codename:	groovy
```

and here is the error:

```
sudo ./displaylink-debian.sh

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.11.0-7612-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/3146?filetype=exe
Do you accept?: [Y/n]: 

Downloading DisplayLink Ubuntu driver:

--2021-04-02 01:56:56--  https://www.synaptics.com/sites/default/files/exe_files/2021-02/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.3.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 52.36.27.145, 52.11.184.137
Connecting to www.synaptics.com (www.synaptics.com)|52.36.27.145|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13889958 (13M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.3.1.zip’

DisplayLink_Ubuntu_5.3.1.zip                    100%[=====================================================================================================>]  13.25M  1.13MB/s    in 14s     

[... 74 lines cut ...]
      |   ^~~~~~~~~~~~~
/home/navid/projects/displaylink-debian/evdi/module/evdi_drv.c:118:19: error: initialization of ‘int (*)(struct drm_gem_object *, struct vm_area_struct *)’ from incompatible pointer type ‘int (*)(struct drm_device *, unsigned int)’ [-Werror=incompatible-pointer-types]
  118 |  .enable_vblank = evdi_enable_vblank,
      |                   ^~~~~~~~~~~~~~~~~~
/home/navid/projects/displaylink-debian/evdi/module/evdi_drv.c:118:19: note: (near initialization for ‘driver.gem_prime_mmap’)
/home/navid/projects/displaylink-debian/evdi/module/evdi_drv.c:119:3: error: ‘struct drm_driver’ has no member named ‘disable_vblank’
  119 |  .disable_vblank = evdi_disable_vblank,
      |   ^~~~~~~~~~~~~~
/home/navid/projects/displaylink-debian/evdi/module/evdi_drv.c:119:20: error: initialization of ‘int (*)(struct drm_file *, struct drm_device *, struct drm_mode_create_dumb *)’ from incompatible pointer type ‘void (*)(struct drm_device *, unsigned int)’ [-Werror=incompatible-pointer-types]
  119 |  .disable_vblank = evdi_disable_vblank,
      |                    ^~~~~~~~~~~~~~~~~~~
/home/navid/projects/displaylink-debian/evdi/module/evdi_drv.c:119:20: note: (near initialization for ‘driver.dumb_create’)
/home/navid/projects/displaylink-debian/evdi/module/evdi_drv.c:119:20: error: initialized field overwritten [-Werror=override-init]
/home/navid/projects/displaylink-debian/evdi/module/evdi_drv.c:119:20: note: (near initialization for ‘driver.dumb_create’)
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:287: /home/navid/projects/displaylink-debian/evdi/module/evdi_drv.o] Error 1
make[2]: *** [Makefile:1834: /home/navid/projects/displaylink-debian/evdi/module] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.11.0-7612-generic'
make[1]: *** [Makefile:77: module] Error 2
make[1]: Leaving directory '/home/navid/projects/displaylink-debian/evdi/module'
make: *** [Makefile:8: all] Error 2
cp: cannot stat 'library/libevdi.so.1.*': No such file or directory
```

Any idea how can I solve this?

i-am-snappy, 2021-04-02:
FYI - i gust recently updated PopOS as well and get same problem. Lets see what eventuates.

Barabazs, 2021-04-02:
Hi! Can you  post the output of `sudo ./displaylink-debian.sh --debug`

ghahramani, 2021-04-02:
Hi,

After digging more, it seems evdi has the problem with kernel 5.11, I could solve the problem by compiling evdi new version and then installed the display link again.

You can follow the instructions below:

1. Uninstall DisplayLink
2. Unplug your dock
3. Uninstall evdi
4. Run the commands mentioned here https://github.com/DisplayLink/evdi/pull/256#issuecomment-780332169
5. Download the latest version of DisplayLink from https://www.displaylink.com/downloads/ubuntu
6. Install the DisplayLink
7. Restart

Now you should have a working monitor :D

varac, 2021-04-06:
> Hi! Can you post the output of sudo ./displaylink-debian.sh --debug

I came across the same error (PopOS 20.10, kernel 5.11.0-7612-generic)

This is the output *before* I tried to reinstall (which failed):

```
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7612-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics (rev 02)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 0 name:Intel

-------------------------------------------------------------------
```

varac, 2021-04-06:
I also noticed that I don't have any `*evdi*` packages installed. Are theses needed ?

```
❯ dpkg -l | grep evdi

❯ apt search evdi
Sorting... Done
Full Text Search... Done
evdi-dkms/groovy,groovy 1.7.0+dfsg-1ubuntu1 all
  Extensible Virtual Display Interface driver kernel module

libevdi0/groovy 1.7.0+dfsg-1ubuntu1 amd64
  Extensible Virtual Display Interface driver client library

libevdi0-dev/groovy,groovy 1.7.0+dfsg-1ubuntu1 all
  EVDI driver client library dev package
```

JeroenGoddijn, 2021-04-07:
> sudo ./displaylink-debian.sh --debug

I have similar output (after uninstalling and having displaylink hub unplugged): 

```
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7612-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0.backup
5.1.26
5.2.14
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: 620
Integrated
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------
```

If I run `git clone https://github.com/AdnanHodzic/displaylink-debian.git && cd displaylink-debian/ && sudo ./displaylink-debian.sh` , I get the following output:

`
DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.11.0-7612-generic is installed
dkms is installed
[... 97 lines cut ...]
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:115:28: note: (near initialization for ‘driver.gem_prime_import’)
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:118:3: error: ‘struct drm_driver’ has no member named ‘enable_vblank’
  118 |  .enable_vblank = evdi_enable_vblank,
      |   ^~~~~~~~~~~~~
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:118:19: error: initialization of ‘int (*)(struct drm_gem_object *, struct vm_area_struct *)’ from incompatible pointer type ‘int (*)(struct drm_device *, unsigned int)’ [-Werror=incompatible-pointer-types]
  118 |  .enable_vblank = evdi_enable_vblank,
      |                   ^~~~~~~~~~~~~~~~~~
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:118:19: note: (near initialization for ‘driver.gem_prime_mmap’)
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:119:3: error: ‘struct drm_driver’ has no member named ‘disable_vblank’
  119 |  .disable_vblank = evdi_disable_vblank,
      |   ^~~~~~~~~~~~~~
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:119:20: error: initialization of ‘int (*)(struct drm_file *, struct drm_device *, struct drm_mode_create_dumb *)’ from incompatible pointer type ‘void (*)(struct drm_device *, unsigned int)’ [-Werror=incompatible-pointer-types]
  119 |  .disable_vblank = evdi_disable_vblank,
      |                    ^~~~~~~~~~~~~~~~~~~
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:119:20: note: (near initialization for ‘driver.dumb_create’)
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:119:20: error: initialized field overwritten [-Werror=override-init]
/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.c:119:20: note: (near initialization for ‘driver.dumb_create’)
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:287: /home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module/evdi_drv.o] Error 1
make[2]: *** [Makefile:1834: /home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.11.0-7612-generic'
make[1]: *** [Makefile:77: module] Error 2
make[1]: Leaving directory '/home/jeroengoddijn/Downloads/displaylink-debian/displaylink-debian/evdi/module'
make: *** [Makefile:8: all] Error 2
cp: cannot stat 'library/libevdi.so.1.*': No such file or directory`

The-Almighty-Sasquatch, 2021-04-11:
I tried this by modifying the files etc myself but can't get the screens to work I have followed the instructions for de-bugging and can't get anywhere. The script half restarted my machine automatically which didn't happen previously.

De-bugging info: (without graphics card info)
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7612-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x17f cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x15d cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x13b cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x119 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Barabazs, 2021-04-12:
Please try again with the latest version from git.

closed 2021-04-12 by Barabazs