AdnanHodzic/displaylink-debian #560: File patching failed during install - driver not installed

IpelaTech, 2021-04-07:
Hi,

Debug Info:

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyssa
Kernel: 5.4.0-58-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): NVIDIA Corporation GF117M [GeForce 610M/710M/810M/820M / GT 620M/625M/630M/720M] (rev a1)
X11 version: 1.20.9-2ubuntu1.2~20.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0xb6 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

**-------------------------------------------------------------------**

After running sudo ./displaylink-debian.sh (both with the git cloned and a downloaded zip), I get the following error:

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.4.0-58-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/3751?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2021-04-07 13:34:14--  https://www.synaptics.com/sites/default/files/exe_files/2021-04/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 52.11.184.137, 52.36.27.145
Connecting to www.synaptics.com (www.synaptics.com)|52.11.184.137|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14501284 (14M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.4.zip’

DisplayLink_Ubuntu_5.4.zip                 100%[======================================================================================>]  13.83M  3.01MB/s    in 6.1s    

2021-04-07 13:34:22 (2.26 MB/s) - ‘DisplayLink_Ubuntu_5.4.zip’ saved [14501284/14501284]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.zip
  inflating: 5.4/displaylink-driver-5.4.0-55.153.run  
Creating directory displaylink-driver-5.4.0-55.153
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.0-55.153  100%  

-------------------------------------------------------------------
Cloning into 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 8 (delta 0), pack-reused 0
Unpacking objects: 100% (46/46), 67.99 KiB | 320.00 KiB/s, done.
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1]: Entering directory '/home/ipeleng/Downloads/displaylink-debian-master/evdi/module'
cat: /etc/redhat-release: No such file or directory
make -C /lib/modules/5.4.0-58-generic/build M=$PWD
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-58-generic'
cat: /etc/redhat-release: No such file or directory
  AR      /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/built-in.a
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_drv.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_modeset.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_connector.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_encoder.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_main.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_fb.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_gem.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_painter.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_params.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_cursor.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_debug.o
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi_ioc32.o
  LD [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi.o
cat: /etc/redhat-release: No such file or directory
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi.mod.o
  LD [M]  /home/ipeleng/Downloads/displaylink-debian-master/evdi/module/evdi.ko
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-58-generic'
make[1]: Leaving directory '/home/ipeleng/Downloads/displaylink-debian-master/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1]: Entering directory '/home/ipeleng/Downloads/displaylink-debian-master/evdi/library'
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.2 -lc -lgcc 
cp libevdi.so.1.7.2 libevdi.so
make[1]: Leaving directory '/home/ipeleng/Downloads/displaylink-debian-master/evdi/library'
patching file 5.4/displaylink-driver-5.4/displaylink-installer.sh
Hunk #1 FAILED at 457.
1 out of 1 hunk FAILED -- saving rejects to file 5.4/displaylink-driver-5.4/displaylink-installer.sh.rej

Thereafter the installation stops and nothing gets installed. Is there something I've done wrong or missed?

Thanks.

Ipeleng

xanthax, 2021-04-08:
I have The Same Issue using a "Lenovo T14 G1 AMD"

Barabazs, 2021-04-12:
Please try again with the latest version from git.

closed 2021-04-12 by Barabazs

IpelaTech, 2021-04-14:
> Please try again with the latest version from git.

Thank you, I will do and get get back to you.

IpelaTech, 2021-04-14:
Hi, installed fine thanks. Service is running, providers are showing, as per the post-install guide but the monitor is still showing as disconnected when I enter the xrandr command. Funny thing is, the device shows up as an audio sink on pulseaudio. I'm going to chalk it up to this being a cheapie no-name brand adapter. Thanks for all the help, though.