AdnanHodzic/displaylink-debian #565: "$cardsub" == "[HD"  is that correct?

Rom888, 2021-04-15:
I have $cardsub=HD on my notebook, and I see "[HD" instead(?) of "HD" in the displaylink-debian.sh:
```
if [ "$cardsub" == "v2/3rd" ] || [ "$cardsub" == "[HD" ] || [ "$cardsub" == "620" ] || [ "$cardsub" == "530" ] || [ "$cardsub" == "540" ] || [ "$cardsub" == "UHD" ];
```
Is that correct?

Barabazs, 2021-04-15:
It looks like a typo, but it's been there since 2018. Can you confirm the script does not work for you?

Rom888, 2021-04-15:
It worked with Ubuntu 19.10 and DisplayLink 5.2.14.
But today it did't work with recently released DisplayLink 5.4 and Ubuntu 20.04.
After "Performing post install steps" only the mouse pointer moved, but the system stops responding.
And I tried to figure out why.

Barabazs, 2021-04-15:
Can you post the output of `sudo ./displaylink-debian.sh --debug`.

If this is your config:
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```
Try replacing it with 
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

Rom888, 2021-04-16:
> Can you post the output of sudo ./displaylink-debian.sh --debug.

Yes, it's below and thanks for help.
Please note that after the system became unresponsive with the displaylink-debian.sh, I installed the DisplayLink software in the official way and manually applied /etc/X11/xorg.conf.d/20-displaylink.conf from the displaylink-debian.sh.
System works, but I have very annoying mouse cursor flicker issue now.

```
$ sudo ./displaylink-debian.sh --debug
-------------------------------------------------------------------
Starting Debug ...
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-71-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: HD
VGA: Intel Corporation HD Graphics 515 (rev 07)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 2 name:modesetting
Provider 1: id: 0x12a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x107 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
-------------------------------------------------------------------
```

Rom888, 2021-04-16:
System works, but I have very annoying mouse cursor flicker issue now with this config too:

```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```

Barabazs, 2021-04-25:
> System works, but I have very annoying mouse cursor flicker issue now with this config too:
> 
This is covered in the post-install steps.
Try the steps mentioned here: https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only

closed 2021-04-25 by Barabazs