AdnanHodzic/displaylink-debian #566: Screen freezes on xrandr settings change

bergben, 2021-04-16:
Hi, thanks for providing this repo! 

I am having post install issues, which are not covered anywhere as far as I could establish. 
When I apply any changes to the xrandr config (e.g. screen layout, resolution, whatever...), the screens just freeze. The only thing that works is adding a second display (just activating it, no other config). 
I tried to apply configs with arandr, xrandr command line, and gnome-control-center. Always the same outcome.

Also, whenever I disconnect the docking station and reconnect it, the screen configuration appears to be random. At least most times the external display does not work anymore, sometimes the other external screen is active etc.
I could not make out a pattern there. I expect this to be related to the other issue though. 

Maybe you have an idea? Thank you!

--------------------------------

Output of debug:

```
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7612-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Graphics
[Quadro
VGA: Intel Corporation UHD Graphics 630 (Mobile)
NVIDIA Corporation GP107GLM [Quadro P1000 Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/imwheel/startup.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x218 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x32c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x30a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2e8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2c6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x248 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 6 associated providers: 1 name:modesetting
```

Barabazs, 2021-04-25:
* Screen freezes when changing settings:
  * I've seen that complaint a lot, but it was never attributed to one specific cause.
  * Might be driver related? Lots of hits on Google.
* Random screen configuration when reconnecting dock:
  * You always have to re-apply your xrandr settings when booting or connecting the dock. That is the normal behaviour.
  * You could look into udev rules for automatically triggering a script that contains your xrandr settings.

closed 2021-04-25 by Barabazs

reynerdev, 2021-08-30:
@bergben did you fixed it ?  After, my laptop freezes, Ubuntu reports this error xf86PlatformDeviceCheckBusID(), and there is an open issue https://gitlab.freedesktop.org/xorg/xserver/-/issues/1151.