AdnanHodzic/displaylink-debian #570: Displays are found but don't receive signal

tomaszgrzegorzn, 2021-05-05:
Hi, 

I'm having an issue with my displays, I'm using a Dell D3100 Docking station, which works fine with with Windows, and used to work fine with my Linux. I've ran through the Post-Install guide, and spend countless houres googling for any solution. I really wanted to avoid a reinstall :) 

I can select and see my displays, but when trying to mark them as active - nothing happens. I've tried to use arandr but I keep getting the following message: "XRandR returned error code 1: b'xrandr: Configure crtc 6 failed \n' "

Thank you in advance for any assitance.

--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7614-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0x139 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x117 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf5 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd3 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

brianor63136, 2021-05-07:
I had the same issue. [This comment](https://github.com/AdnanHodzic/displaylink-debian/issues/206#issuecomment-458619074) got my displays working. But now, if I try to change their position in XRandR or ARandR, my machine freezes.

This doesn't happen when I try to do the same thing with a HDMI directly connected to my machine.
The same behaviour happens if I remove the USB unexpectedly.

My debug information:
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7614-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1028:09e1]
GTX
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation TU117M [GeForce GTX 1650 Ti Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 1 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x31c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2fa cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2d7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e3 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting

tomaszgrzegorzn, 2021-05-07:
Many thanks @brianor63136 - after updating my system with newest EVDI, and the changes under the link you provided, it works like a charm. What's more, I can unplug the USB or utilize arandr to rearrange my displays without the freezes you have mentioned. Still wondering if it's a workaround, or permanent solution, if there are no noticable side effects.

brianor63136, 2021-05-07:
Glad it's up and running @tomaszgrzegorzn

I'll make a new issue for my crash problem so that you can close this off.

tomaszgrzegorzn, 2021-05-07:
Thanks once again, closing it now.

closed 2021-05-07 by tomaszgrzegorzn