AdnanHodzic/displaylink-debian #577: Blank screen -  Nvidia / Intel combo

zach-is-my-name, 2021-05-23:
Hello! Thank you, THANK YOU for crafting this script and the support you've provided everyone on this most critical computer interface functionality.  

I'm experiencing a blank screen with cursor - I've read just about every pertinent issue and applied just about every config setting.  Please let me know if I'm missing something...

```--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-73-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nouveau
Subsystem: Device
Device
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation Device 1f95 (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.2
X11 configs: /etc/X11/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x43 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 1 associated providers: 5 name:modesetting
Provider 1: id: 0x15d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x13b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x119 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xf7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0xd5 cap: 0x2, Sink Output crtcs: 4 outputs: 3 associated providers: 1 name:modesetting
```

Barabazs, 2021-05-24:
Please delete the `/etc/X11/xorg.conf` file and replace your `/etc/X11/xorg.conf.d/20-displaylink.conf` config with:
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

zach-is-my-name, 2021-05-24:
No effect

```Release: focal
Kernel: 5.4.0-73-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nouveau
Subsystem: Device
Device
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation Device 1f95 (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x43 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 1 associated providers: 5 name:modesetting
Provider 1: id: 0x15d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x13b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x119 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xf7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0xd5 cap: 0x2, Sink Output crtcs: 4 outputs: 3 associated providers: 1 name:modesetting
```

mosqueteiro, 2021-06-01:
Nvidia/Intel combo here too. Experiencing similar blank screen on my second display even though it is detected.

```
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7614-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
X11 version: 1.20.9-2ubuntu1.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
	Identifier "DisplayLink"
	Driver     "modesetting"
	Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 4 name:modesetting
Provider 1: id: 0x16c cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x14a cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x128 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xf3 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0xd1 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

```

Barabazs, 2021-06-03:
@ mosqueteiro Your last provider is not associated. Maybe that resolves your issue. Please open a separate issue if it still doesn't work.

@zach-is-my-name What is the output of `service displaylink status`?

closed 2021-06-19 by Barabazs