AdnanHodzic/displaylink-debian #578: Display link doesn't associate the providers with amdgpu

n1mh, 2021-05-23:
Hi,

I've followed the readme.md and postinstall.md file throughputly and it fails **only** with AMD videocard. Both laptops has debian buster (updated), lightdm and mate window manager. I've installed `displaylink-debian.sh` from the scratch and after purge every packege involved.

In the AMD laptop I'm using `buster-backports` repo to have an updated version of `amdgpu` and few other packages. The installation commands are:

```
sudo apt install -t buster-backports -y linux-headers-5.10.0-0.bpo.4-amd64 linux-image-5.10.0-0.bpo.4-amd64 linux-support-5.10.0-0.bpo.4 linux-kbuild-5.10
sudo apt install -t buster-backports -y linux-config-5.10 linux-source-5.10 libc6-dev linux-libc-dev libc-dev-bin dkms libelf-dev                                                                               
sudo apt install -t buster-backports -y libdrm-dev libdrm2 libdrm-amdgpu1
```

```
$ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.10.0-0.bpo.4-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Renoir
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Renoir (rev c6)
VGA (3D): 
X11 version: 1.20.4-1+deb10u3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No existe el fichero o el directorio
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcd cap: 0x0 crtcs: 4 outputs: 2 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

I've moved `/etc/X11/xorg.conf.d/20-displaylink.conf` because if it existed, xrand didn't detect any provider.

At this point, everything seems to be ready:

```
$ xrandr --listproviders
Providers: number : 5
Provider 0: id: 0xcd cap: 0x0 crtcs: 4 outputs: 2 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

But when I try to associate the providers it fails:

```
$ xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0xcd
  Serial number of failed request:  19
  Current serial number in output stream:  20
```

I've tried with two different displaylink to get the same error.

Any idea what's going on?

Thanks for your fantastic work!

n1mh, 2021-05-23:
To be sure that I was deactivating Secure Boot, I've installed it on a third laptop with Intel graphic card and the association with the providers worked fine.

So now I pretty confident that whatever it is, it is related with `amdgpu`.

Barabazs, 2021-05-24:
There are multiple issues/bugs with amdgpu: https://github.com/DisplayLink/evdi/issues?q=is%3Aissue+is%3Aopen+amdgpu
Maybe you can try with the modesetting driver while the issues are resolved upstream.
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

n1mh, 2021-05-24:
> There are multiple issues/bugs with amdgpu: https://github.com/DisplayLink/evdi/issues?q=is%3Aissue+is%3Aopen+amdgpu
> Maybe you can try with the modesetting driver while the issues are resolved [...]

Thanks for the link, I'll follow the issues in evdi hoping that something change about amdgpu :).

I'll test that configuration in few hours and let you know what happens.

thanks!

n1mh, 2021-05-25:
That configuration didn't work, it shows the same error.

I will wait for a new, open, fully-complaint driver for amdgpu that supports displaylink then. Maybe in the next century :D.

Thank you all for this great script!

closed 2021-05-25 by n1mh