AdnanHodzic/displaylink-debian #581: Wrong Kernel Source

cement-head, 2021-05-24:
Trying to use on a System76 machine.

Script won't recognise linux-source-5.11; wants to install linux-source & linux-source-5.4

Barabazs, 2021-05-27:
Can you post the output of the install please?

cement-head, 2021-06-01:
Nope, locks up and crashes. No output generated.

Barabazs, 2021-06-03:
I'm not sure if this is related to how displaylink-debian tries to install dependencies.
All we do is specify the packages, without any version numbers.

I would appreciate it if anyone can submit install logs or additional details.

ryepup, 2021-07-15:
I also have a System76, and working through display link troubles. I'm uninstalling some stuff to get to a "clean" state then will get an install log.

One possible clue; looks like the default `linux-source` package is 5.4:

```console
$ sudo apt update
...
$ apt search linux-source
Sorting... Done
Full Text Search... Done
linux-source/focal-security,focal-updates,focal-updates 5.4.0.77.80 all
  Linux kernel source with Ubuntu patches

linux-source-5.11.0/focal,focal,focal 5.11.0-7620.21~1624379747~20.04~3abeff8~dev all
  Linux kernel source for version 5.11.0 with Ubuntu patches

linux-source-5.4.0/focal-security,focal-updates,focal-updates,now 5.4.0-77.86 all [installed,automatic]
  Linux kernel source for version 5.4.0 with Ubuntu patches
$ apt show linux-source
Package: linux-source
Version: 5.4.0.77.80
Priority: optional
Section: devel
Source: linux-meta
Origin: Ubuntu
Maintainer: Ubuntu Kernel Team <kernel-team@lists.ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 18.4 kB
Depends: linux-source-5.4.0
Download-Size: 2,444 B
APT-Sources: http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages
Description: Linux kernel source with Ubuntu patches
 This package will always depend on the latest Linux kernel source code
 available. The Ubuntu patches have been applied.

```

ryepup, 2021-07-15:
Ok, got an log of the install using [`script`](https://man7.org/linux/man-pages/man1/script.1.html). Removing some progress bar noise and `script`'s formatting I've got:

```text
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.11.0-7620-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/3751?filetype=exe
Do you accept?: [Y/n]: 

Downloading DisplayLink Ubuntu driver:

--2021-07-15 19:12:55--  https://www.synaptics.com/sites/default/files/exe_files/2021-04/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.4-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 52.39.36.8, 34.210.13.121
Connecting to www.synaptics.com (www.synaptics.com)|52.39.36.8|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 14501284 (14M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.4.zip’

<snip download progress>  
DisplayLink_Ubuntu_ 100%[===================>]  13.83M  1.42MB/s    in 8.0s    

2021-07-15 19:13:03 (1.74 MB/s) - ‘DisplayLink_Ubuntu_5.4.zip’ saved [14501284/14501284]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.4.zip
  inflating: 5.4/displaylink-driver-5.4.0-55.153.run  
Creating directory displaylink-driver-5.4.0-55.153
Verifying archive integrity...    100%   All good.
Uncompressing DisplayLink Linux Driver 5.4.0-55.153     100%  
patching file 5.4/displaylink-driver-5.4/displaylink-installer.sh
Hunk #1 succeeded at 456 (offset -1 lines).
[... 124 lines cut ...]
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 5 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x30e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2ec cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2ca cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e5 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 1 associated providers: 1 name:modesetting
```

Ideas:

* maybe the compilation step gets weird if I'm running a 5.11 kernel with a 5.4 linux-source?
* maybe System76 has a weird setup for their nvidia cards? I have an nvidia card but the `drv_nvidia` check isn't finding it 

```console
$ lspci | grep -i nv
01:00.0 VGA compatible controller: NVIDIA Corporation TU106 [GeForce RTX 2060] (rev a1)
01:00.1 Audio device: NVIDIA Corporation TU106 High Definition Audio Controller (rev a1)
01:00.2 USB controller: NVIDIA Corporation TU106 USB 3.1 Host Controller (rev a1)
01:00.3 Serial bus controller [0c80]: NVIDIA Corporation TU106 USB Type-C UCSI Controller (rev a1)
```

ryepup, 2021-07-16:
I tried uninstalling and then changing the script to identify my nvidia card:

```diff
 drv=$(lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g')
-drv_nvidia=$(lspci | grep -i '3d controller' | sed 's/^.*: //' | awk '{print $1}')
+drv_nvidia=$(lspci | grep -i 'RTX' | sed 's/^.*: //' | awk '{print $1}')
 cardsub=$(lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5)
 ```

Install went basically the same as above, but my `/etc/X11/xorg.conf.d/20-displaylink.conf` now has all the nvidia settings.

I'm still getting freezes when I change monitor settings, going to try uninstalling/reinstalling with `linux-source-5.11` instead of 5.4

ryepup, 2021-07-16:
I tried using a kernel-specific version of `linux-source`, no luck:

```diff
 else
-       deps=(unzip linux-headers-$(uname -r) dkms lsb-release linux-source x11-xserver-utils wget libdrm-dev libelf-dev git pciutils)
+       deps=(unzip linux-headers-$(uname -r) dkms lsb-release linux-source-$(uname -r | cut -d'-' -f 1) x11-xserver-utils wget libdrm-dev libelf-dev git pciutils)
 fi
```
The install log was basically the same as above, but installing `linux-source-5.11.0` instead.

ryepup, 2021-07-17:
made a separate issue for the nvidia detection and put up a PR: #596

I think the root problem here is the upstream `linux-source` package hasn't been updated right. I'm unclear of the impact; I see no difference in behavior when I use `linux-source-5.11.0` instead of `linux-source-5.4.0`, but I've got other issues conflating the matter.

Best guess is this is an issue for https://packages.ubuntu.com/focal/linux-source, not displaylink-debian

closed 2021-12-21 by Barabazs