AdnanHodzic/displaylink-debian #582: Computer seems to detect screens but can't "turn them on" in gnome displays properly

The-Almighty-Sasquatch, 2021-06-02:
Not terribly familiar with all this display command line stuff but desperately trying to get my dock to work with my pop os 20.10 computer.

It seems something is missing from the install under the xorg.config. When i initially run the displaylink-debian script it doesn't prompt for a reboot instead it immediately "semi reboots" i.e. doesn't go to the screen where i have to decrypt my drives just straight to user login.

in gnome displays i can see my screens and arrange them but toggling the "enable" icon doesn't seem to work at all

Here is debug information:
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7614-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1028:09e1]
RTX
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x18c cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x16a cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x147 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x119 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Hope you can help, and thanks for taking the time and effort to make this in the first place!
Thanks again,
James

The-Almighty-Sasquatch, 2021-06-02:
I think this might be something similar to https://github.com/AdnanHodzic/displaylink-debian/issues/581

The-Almighty-Sasquatch, 2021-06-02:
not sure though

Barabazs, 2021-06-03:
Did you try anything else before installing with `displaylink-debian.sh`?
Install with the official package or another one?

The-Almighty-Sasquatch, 2021-06-06:
I had previously and then it stopped working after an update, so I tried I uninstalled everything but perhaps this wasn't enough? I will try and find all packages and residual files and uninstall everything. Do you have the location for anything which could have been missed?

The-Almighty-Sasquatch, 2021-06-06:
Hi,
I have now tried removing everything with displaylink in it and ran uninstall scripts for both the official and displaylink-debian. Still have same issue.
Thanks,
James

Barabazs, 2021-06-06:
Please repost the output of `sudo ./displaylink-debian.sh --debug`

The-Almighty-Sasquatch, 2021-06-06:
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7614-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1028:09e1]
RTX
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x18c cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x16a cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x147 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x119 cap: 0x6, Sink Output, Source Offload crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
Thanks,
James

Barabazs, 2021-06-06:
How did you reinstall it this time?

The-Almighty-Sasquatch, 2021-06-06:
as per instructions on here using git

Barabazs, 2021-06-06:
OK, thanks for confirming. Are you running wayland or Xorg?
Please try this config for `/etc/X11/xorg.conf.d/20-displaylink.conf`
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
Don't forget to sign out and re-associate the providers etc. Please also post a new debug output after you're done.

The-Almighty-Sasquatch, 2021-06-06:
Thanks for your help,
So I created that file as nothing existed in that folder (sudo nano /etc/X11/xorg.conf.d/20-displaylink.conf) and then reconfigured Xorg (sudo dpkg-reconfigure xserver-xorg), logged out and it all seems to work now!
Cheers
James

The-Almighty-Sasquatch, 2021-06-06:
--------------- Linux system info ----------------

Distro: Pop
Release: groovy
Kernel: 5.11.0-7614-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1028:09e1]
RTX
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x31c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2fa cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2d7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e7 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting

-------------------------------------------------------------------

This is new working debug

The-Almighty-Sasquatch, 2021-07-13:
Actually doing this didn't totally solve my issues, I've only just around to typing this but it's been an issue since I last closed this, (i.e. before i upgraded to hirsute. I can see all the screens but every time I try and rearrange the screens arrangements either via command line or gnome displays, all the displays freeze including the internal laptop display.

Here is the debug information:
--------------- Linux system info ----------------

Distro: Pop
Release: hirsute
Kernel: 5.11.0-7620-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1028:09e1]
RTX
VGA: Intel Corporation CometLake-H GT2 [UHD Graphics] (rev 05)
NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x30f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2ed cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2cb cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e7 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting

-------------------------------------------------------------------

ryepup, 2021-07-16:
I'm seeing the same freezes as @The-Almighty-Sasquatch, and might be on similar hardware. I have also been adding (spamming?) #581 with my debugging efforts.

```text
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.11.0-7620-generic

---------------- DisplayLink info ----------------

Driver version: 1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1558:50d3]
[1558:50d3]
VGA: Intel Corporation UHD Graphics (rev 05)
NVIDIA Corporation TU106 [GeForce RTX 2060] (rev a1)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 4 outputs: 5 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x313 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2f1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2cf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2a8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1e5 cap: 0x2, Sink Output crtcs: 3 outputs: 1 associated providers: 1 name:modesetting

```

ryepup, 2021-07-16:
I wonder if this is related to https://bugs.launchpad.net/ubuntu/+source/xorg-server/+bug/1931547 or https://github.com/DisplayLink/evdi/issues/258

closed 2021-12-21 by Barabazs