AdnanHodzic/displaylink-debian #583: After install displaylink all screens shows green boxes

simonppg, 2021-06-03:
I succesfully installed displaylink but now all my screens look shows green boxes or lines, can anybody take a looks at my config and tell me some clue, I have no idea where to look to fix this, I'm not sure if this is a displaylink issue or something else.

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.8.0-53-generic
![Screenshot from 2021-06-02 17-27-36](https://user-images.githubusercontent.com/3270997/120568088-0f26ab00-c3c8-11eb-8c95-9f991ed8b595.png)

---------------- DisplayLink info ----------------

Driver version: 1.7.0+dfsg
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Iris Plus Graphics G7 (rev 07)
VGA (3D): 
X11 version: 1.20.9-2ubuntu1.2~20.04.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 4 name:Intel
Provider 1: id: 0xef cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xcd cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xab cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x6f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```
[image]

Barabazs, 2021-06-03:
Did you add additional settings to your `20-displaylink.conf`?

Try with only this please:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

simonppg, 2021-06-03:
I just added this two, because it didn't work before

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

I'll try what you suggested, thanks

simonppg, 2021-06-03:
@Barabazs that works perfect, thank you

closed 2021-06-03 by simonppg