AdnanHodzic/displaylink-debian #584: Mageia support

Loulou-F, 2021-06-08:
Hi,

Do you plan to add a Mageia support ? I would definitely grant you a few coins for that. When running displaylink-debian.sh from Mageia 8, it says "This is a Redhat based distro...", which, as long as I know, is false.

Here is the output of the --debug option :

-----
--------------- Linux system info ----------------

Distro: Mageia
Release: mga8
Kernel: 5.10.41-desktop-1.mga8

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [1558:6588]
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): 
X11 version: 
X11 configs: /etc/X11/gdm/custom.conf /etc/X11/xinit/xinput.d/im-cedilla.conf /etc/X11/xorg.conf /etc/X11/xorg.conf.d/00-keyboard.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
-----

Thank you for your time and congrats for your efforts, 
Louis

Barabazs, 2021-06-08:
Hi @Loulou-F can you give me the output `cat /etc/redhat-release ` please?
displaylink-debian.sh performs a check to see if that file exists and then exits.

Barabazs, 2021-06-08:
It looks like your distro is rpm based. You should try [displaylink-rpm](https://github.com/displaylink-rpm/displaylink-rpm).
Please let me know how it went.

Loulou-F, 2021-06-08:
It installed but it is still not working.

closed 2021-06-19 by Barabazs